luxonis/depthai #769: [QUESTION] {setIrFloodLightBrightness}

stephanschulz, 2022-08-13:
do you have example code showing how to `setIrFloodLightBrightness` and 'setIrLaserDotProjectorBrightness' for the OAK-D Pro POE?

I am specifically working from this blaze pose repo
https://github.com/geaxgx/depthai_blazepose/blob/main/BlazeposeDepthaiEdge.py#L104

Thanks.

Erol444, 2022-08-13:
Hi @stephanschulz ,
Could you share the screenshot? Also, note that the color camera has IR filter, only stereo cameras have IR notch filter on them (so they "see" IR light from illumination LED).
Thanks, Erik

stephanschulz, 2022-08-13:
Here is the main file where i am trying to turn the LED and/or laser on
https://gist.github.com/stephanschulz/d710a77681e4c15c34ad8484821c1354#file-blazeposedepthaiedge_osc-py-L226

Here a screen shot of the terminal print outs:
[image]

stephanschulz, 2022-08-13:
Here a screen shot showing the blaze pose python script sending OSC messages to my custom c++ app
[image]

Erol444, 2022-08-13:
Could you share a mono camera image? As mentioned, color camera doesn't perceive IR light from illumination LED.

stephanschulz, 2022-08-13:
Sorry about that.
I now understand. It took me a minute to learn how to display the mono cameras.
Here I am showing the right camera and it looks like laser and IR Led are working.
[image]

I followed some of the code here:
https://docs.luxonis.com/projects/api/en/latest/samples/VideoEncoder/rgb_mono_encoding/

Thanks again.