luxonis/depthai #770: [QUESTION] run only on OKA-D Pro POE as standalone device

stephanschulz, 2022-08-14:
Hello.

I was wondering if you have a. tutorial explaining how i can make any of my code run only on my OAK-D Pro POE **without the need of a terminal window** being open.
Specifically i am working successfully with this repo https://github.com/geaxgx/depthai_blazepose

Right now i cd to my project repo and call python3.8 demo_osc.py -e --oscIP 10.100.0.101 --oscPort 12345 -xyz
I have modified the code already to not show a live video, not show any window other than the terminal window.
I added code to send the skeleton data over UPD / OSC.

I am hoping that i can upload the code, then disconnect my computer and have the OAK-D send the OSC messages in to the network where an other computer can pick it up.

Thanks for your help.

Erol444, 2022-08-15:
Hi @stephanschulz ,
You should check our [Standalone mode documentation page](https://docs.luxonis.com/projects/api/en/latest/tutorials/standalone_mode/), which addresses just that. Thoughts?
Thanks, Erik

stephanschulz, 2022-08-15:
thanks for the link.
In my case i am uncertain if i need to wrap my OSC sending code in to a pipeline and if so how to do that correctly.
https://github.com/stephanschulz/depthai_blazepose/blob/main/depthai_osc/oscSender_osc.py#L103