luxonis/depthai #776: Converting tensorflow model to blob

tarekmuallim, 2022-08-24:
Hi,
I am trying to convert a model with input shape [1,224,224,3].
But I am getting a blob with input shape [3,224,244,1] which is not suitable to pass camera output to.
Is there a way to convert the model and keep the original input shape?

brmarkus, 2022-08-24:
Can you provide details about how you are trying to convert the model, with source-code, with scripts using which command-line?

Erol444, 2022-08-24:
@tarekmuallim what do you mean by ` is not suitable to pass camera output to.`? Most (if not all) models from Open-Model-Zoo and DepthAI-Model-Zoo are in planar representation (which is what you have), you just need to change `camRgb.setInterleaved(False)` for your ColorCamera. Thoughts?
Thanks, Erik

tarekmuallim, 2022-08-25:
Hi @Erol444 
I am still new to depthai, I didn't know before that I can get planar output from the rgb camera. Thank you for mentioning that. However I am still wondering how to make the conversion without changing the input shape order.
Another question came to my mind : can I apply transpose to the output of the rgb camera.

tarekmuallim, 2022-08-25:
> Can you provide details about how you are trying to convert the model, with source-code, with scripts using which command-line?

Hi @brmarkus
I am using:
`mo --saved_model_dir /tensorflow_model --output_dir /openvino_model --input_shape [1,224,224,3] --data_type FP16 --scale 255 --reverse_input_channels`

`/home/tem/intel/openvino_2022/tools/compile_tool/compile_tool -m /openvino_model/saved_model.xml -ip U8 -op FP16 -d MYRIAD -VPU_NUMBER_OF_SHAVES 6 -VPU_NUMBER_OF_CMX_SLICES 6 -o /blob_model/model_u.blob`

closed 2022-08-31 by tarekmuallim