luxonis/depthai #781: [BUG] {OAK-D Pro POE with depthai_demo.py says No DepthAI device found!}

rjuliaros, 2022-08-31:
**Describe the bug**
I have followed the steps mentioned in the [installation guide](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html). When I try to run the depthai_demo.py it always says No DepthAI device found!, the gui opened but I am not able to see anything about the camera. I am using a PC with OS: Ubuntu 20-04 I have already made the same experiment with Ubuntu 18.04 with identical result. With the same set up I have ran the python code in Windows with any problem. I can ping the camera in the 169.254.1.222 and my IP is 169.254.1.10. Have you hear before about it? 

**Expected behavior**
The image should be visible as in the Windows case.

**Screenshots**
[image]

**Attach system log**
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.8.0-43-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default Jun 22 2022 20:18:18",
    "python_compiler": "GCC 9.4.0",
    "python_implementation": "CPython",
    "python_version": "3.8.10",
    "release": "5.8.0-43-generic",
    "system": "Linux",
    "version": "#49 ~ 20.04.1-Ubuntu SMP Fri Feb 5 09:57:56 UTC 2021",
    "win32_ver": "",
    "packages": [
        "actionlib==1.13.2",
        "angles==1.9.13",
        "apturl==0.5.2",
        "argon2-cffi==21.3.0",
        "argon2-cffi-bindings==21.2.0",
        "asttokens==2.0.8",
        "attrs==22.1.0",
        "backcall==0.2.0",
        "bcrypt==3.1.7",
        "beautifulsoup4==4.11.1",
        "bleach==5.0.1",
        "blinker==1.4",
        "blobconverter==1.3.0",
        "bondpy==1.8.6",
        "boto3==1.24.63",
        "botocore==1.27.63",
        "breezy==3.0.2",
        "Brlapi==0.7.0",
        "camera-calibration-parsers==1.12.0",
        "catkin==0.8.10",
        "catkin-pkg==0.5.2",
        "catkin-pkg-modules==0.5.2",
        "catkin-tools==0.9.0",
        "certifi==2019.11.28",
        "cffi==1.15.1",
        "chardet==3.0.4",
        "charset-normalizer==2.0.12",
        "Click==7.0",
        "colorama==0.4.3",
        "command-not-found==0.3",
        "configobj==5.0.6",
        "cryptography==2.8",
        "cupshelpers==1.0",
        "cv-bridge==1.16.0",
        "cycler==0.11.0",
        "dbus-python==1.2.16",
        "debugpy==1.6.3",
        "decorator==5.1.1",
        "defer==1.0.6",
        "defusedxml==0.7.1",
        "Deprecated==1.2.7",
        "depthai==2.17.3.0",
        "-e git+https://github.com/luxonis/depthai.git@41b95a9e225562fcbb4f5815b0767afa2167d79d#egg=depthai_sdk&subdirectory=depthai_sdk",
        "diagnostic-analysis==1.11.0",
        "diagnostic-common-diagnostics==1.11.0",
        "diagnostic-updater==1.11.0",
[... 256 lines cut ...]
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "SuperPlus"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux rjulia 5.8.0-43-generic #49~20.04.1-Ubuntu SMP Fri Feb 5 09:57:56 UTC 2021 x86_64 x86_64"
    ]
}
System info gathered successfully - saved as "log_system_information.json"

**Additional context**
--

Erol444, 2022-09-05:
Hi @rjuliaros ,
I apologize for the delay. Have you checked [getting started with POE docs](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html)? Another option is to specify the IP using `python depthai_demo.py -dev 169.254.1.222`.
Thanks, Erik

YoshiyasuIzumi, 2022-11-09:
@Erol444 
I have similar issue and I tried `python depthai_demo.py -dev 169.254.1.222`.
I also success to ping to "169.254.1.222" by setting my laptop static address to "169.254.1.10".
But after running `python depthai_demo.py -dev 169.254.1.222`. I got following error.
```
python depthai_demo.py -dev 169.254.1.222
Using depthai module from:  /home/yi/.local/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
Depthai version installed:  2.19.0.0
Setting up demo...
  File "/home/yi/repo/pg/camera_evaluation/depthai/depthai_demo.py", line 655, in run
    self.instance.run_all(self.conf)
  File "/home/yi/repo/pg/camera_evaluation/depthai/depthai_demo.py", line 131, in run_all
    self.setup(conf)
  File "/home/yi/repo/pg/camera_evaluation/depthai/depthai_demo.py", line 179, in setup
    self._deviceInfo = getDeviceInfo(self._conf.args.deviceId, args.debug)
  File "/home/yi/repo/pg/camera_evaluation/depthai/depthai_sdk/src/depthai_sdk/utils.py", line 220, in getDeviceInfo
    raise RuntimeError("No DepthAI device found!")
RuntimeError: No DepthAI device found!
```
What's going wrong?

rjuliaros, 2022-11-11:
Hi all,

I managed to make it work by using an old version of depthai, I'm not sure
about the origin of the issue. I will check it in a couple of weeks, I will
write again If I find something.

Kind regards,

-- 
<https://www.robotnik.eu>
Raquel Juliá
R&D DEPARTMENT
(+34) 96 147 54 00
***@***.*** | robotnik.eu <https://www.robotnik.eu>
<https://www.facebook.com/Robotnik.Automation/?notif_t=page_admin&notif_id=1506077317259585>
  <https://twitter.com/RobotnikRobots>
<https://www.linkedin.com/company/robotnik-automation-sll/?viewAsMember=true>
  <https://www.instagram.com/robotnik.automation/>

-- 

Please note that this message may contain confidential information. If you 
have received this message by mistake,please inform the sender of the 
mistake, then delete the message from your system without making, 
distributing
or retaining any copies of it. Although we believe that the 
message and any attachments are free from viruses and
other errors that 
might affect the computer or IT system where it is received and read, the 
recipient opens the
message at his or her own risk. We assume no 
responsibility for any loss or damage arising from the receipt or use
of 
this message.
If Robotnik Automation, SLL processes personal data relating 
to physical persons, such processing will meet the
requirements of 
applicable data protection legislation.Please see our Privacy Policy 
<https://robotnik.eu/privacy-policy/> and our Legal Notice 
<https://robotnik.eu/legal-notice/>.

themarpe, 2022-11-11:
@YoshiyasuIzumi @rjuliaros 

We've got a fix for this already in the works (`fw_poe_startup_bugfix` in core) - we are targeting 2.19.1 to release this bugfix. 
Thanks for the report!

closed 2022-11-11 by themarpe