luxonis/depthai #782: [Feature-Request] Multi-cam image stitching

Erol444, 2022-09-01:
### Start with the `why`:
Because it's simples/cheaper to use multiple camera sensors (eg. using [our Modular Camera design](https://docs.luxonis.com/projects/hardware/en/latest/#modular-camera-designs)) that are at an angle compared to using a high-res high-fov camera sensor. Also for a full 360deg view you need multiple camera sensors, and then stitch the image together.

### Move to the `what`:
Create a demo for multi-camera stitching. We would first need multi-cam extrinsics calibration script in order to then develop image stitching.

### Move to the `how`:
Computer vision.

tinito, 2023-03-30:
Any progress on the stitching part?

Erol444, 2023-03-30:
Hi @tinito , we haven't yet created a demo of this, as there wasn't much interest.

john-maidbot, 2023-04-21:
I was also interested in this. However, in my first attempt, I hit a dead end trying to convert the kornia image stitching to an onnx model (I even tried different matchers and feature descriptors. All failed to convert). So many errors and warnings from the onnx tracer... And I didn't want to go through trying to patch all that or rewrite the kornia models.
 So I am going to try to see if I can at least offload the feature extraction to the device. And then do matching and homography estimation on the host.