luxonis/depthai #783: [Feature-Request] Multi-cam calibration

Erol444, 2022-09-01:
### Start with the `why`:
We first need multi-cam extrinsics calibration script in order to develop demos like:
- [Multi-cam pointcloud alignment/fusion](https://github.com/luxonis/depthai/issues/747)
- [Multi-cam image stitching](https://github.com/luxonis/depthai/issues/782)
- [Multiple OAK cam stereo vision](https://github.com/luxonis/depthai/issues/766)

### Move to the `what`:
Develop a calibration script that would estimate camera extrinsics (translation/orientation of one cam based to the other). I believe our [calibration.py script](https://github.com/luxonis/depthai/blob/main/calibrate.py) already does something similar with the help of charuco board, so that would be worth investigating. Also investigate using target-free calibration, eg. using [LoFTR](https://zju3dv.github.io/loftr/) and using multiple sensors on the OAK camera.

Erol444, 2022-10-10:
Done: [Multi-cam calibration](https://github.com/luxonis/depthai-experiments/tree/master/gen2-multiple-devices/multi-cam-calibration#multi-camera-calibration)

closed 2022-10-10 by Erol444

JarlLemmens, 2023-09-22:
hey, when I try to run this code, I am able to start up the cameras, and I can select which camera with the keys. But as soon as I press 'p', the script gets stuck in the 'capture still image' loop. I tried different positions of the checkerboard en made sure nothing was moving in the background, but without luck. Is there anything that could be going wrong there?

Erol444, 2023-09-22:
Hi @JarlLemmens , could you provide any output logs from the terminal?

JarlLemmens, 2023-09-22:
@Erol444 
`Found 2 devices
[1844301051E6620F00] [10.11.0.73] [4.931] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
=== Connected to 1844301051E6620F00
[184430105130630F00] [10.11.0.72] [4.901] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
=== Connected to 184430105130630F00
Selected camera 1
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
`

JarlLemmens, 2023-09-22:
@Erol444 

Found 2 devices
[1844301051E6620F00] [10.11.0.73] [4.931] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
=== Connected to 1844301051E6620F00
[184430105130630F00] [10.11.0.72] [4.901] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
=== Connected to 184430105130630F00
Selected camera 1
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still
did not recieve still image - retrying
capturing still

and this goes on indefinitely

Erol444, 2023-09-22:
Hi @JarlLemmens , I just tested it, and it works as expected for me, both USB/POE, IMX378 and OV9782... Could you capture a video of this?

JarlLemmens, 2023-09-22:
https://github.com/luxonis/depthai/assets/64595463/70e43015-f001-46c5-9352-d5c46ba441df

JarlLemmens, 2023-09-22:
[capturing-image-still.zip](https://github.com/luxonis/depthai/files/12702070/capturing-image-still.zip)

uploading the video directly said it was corrupted, so hopefully it works like this

Erol444, 2023-09-22:
Hi @JarlLemmens , that is strange.. Have you changed the code at all? If not, could you try running [RGB camera control example](https://docs.luxonis.com/projects/api/en/latest/samples/ColorCamera/rgb_camera_control/#rgb-camera-control) and try pressing `c` to capture still image, and check whether you get the still image back to the computer?

JarlLemmens, 2023-09-22:
@Erol444 nope, I didn't change anything, just the config. I will try this monday morning, thanks

Erol444, 2023-09-22:
So looking at the code, it might be that you never get the still frame within 1 sec, and then it just clears queue. Could you [check your bandwidth](https://github.com/luxonis/depthai-experiments/blob/master/random-scripts/oak_bandwidth_test.py), [latency](https://github.com/luxonis/depthai-experiments/blob/master/random-scripts/oak_latency_test.py), and do [POE test](https://github.com/luxonis/depthai-experiments/blob/master/random-scripts/poe_test.py)?

Erol444, 2023-09-22:
Hi @JarlLemmens , then that would be the culrpit of the issue. Could you change the switch with something that would work normally? As otherwise you'll have a bunch of issues with messages being dropped down the road.

JarlLemmens, 2023-09-22:
@Erol444 at the moment this is our only option while we're looking into better options.. would there be a way to calibrate from images?

Erol444, 2023-09-22:
@JarlLemmens  one option would be to rather take `.isp` output and constantly stream it to the host, isntead of sending still frames back. Then when user clicks on `p`, you just take latest frame that arrived to the host (as many of them will just be dropped). Perhaps also implement video encoding (MJPEG) to preserve bandiwdth (it's already used for still frames).

JarlLemmens, 2023-09-22:
@Erol444 How would I approach to implement this latest frame idea? If I make sure nothing moves in the frames, this shouldnt be an issue I'd say. We're currently only interested in getting the calibration mapping from camera to camera.

Erol444, 2023-09-22:
so instead of linking camRgb.still to videoEnc, link camRgb.isp. Then you should always receive frames in the queue, and then you can remove sending still message, and also remove that tryGetAll() to clear messages, as you just want to receive latest. When creating queue you should have it as nonblocking and size=1 (as [per docs here](https://docs.luxonis.com/projects/api/en/latest/components/device/#specifying-arguments-for-getoutputqueue-method)).

JarlLemmens, 2023-09-25:
@Erol444 Soo, I tried to change the things you mentioned, just to confirm I did the right steps:

`self.still_queue = self.device.getOutputQueue(name="still", maxSize=1, blocking=False)` in line 17 of the camera.py file

I kept this the same, as it is size 1 and nonblocking.

in line 64, I changed

`cam_rgb.still.link(still_encoder.input)` to `cam_rgb.isp.link(still_encoder.input)`

in line 89, I commented out `self.still_queue.tryGetAll()`. I tried both without this line at all or changing it to `self.still_queue.tryGet()`

In both cases I get an output that says

[1844301051E6620F00] [10.11.0.73] [5.131] [VideoEncoder(2)] [warning] Arrived frame type (2) is not either NV12 or YUV400p (8-bit Gray)
[1844301051E6620F00] [10.11.0.73] [5.161] [VideoEncoder(2)] [warning] Arrived frame type (2) is not either NV12 or YUV400p (8-bit Gray)
[1844301051E6620F00] [10.11.0.73] [5.194] [VideoEncoder(2)] [warning] Arrived frame type (2) is not either NV12 or YUV400p (8-bit Gray)
[1844301051E6620F00] [10.11.0.73] [5.227] [VideoEncoder(2)] [warning] Arrived frame type (2) is not either NV12 or YUV400p (8-bit Gray)

etc etc

and the capturing image still issue continues to be the same. Should I change something else anywhere as well?

JarlLemmens, 2023-09-25:
@Erol444 also, I checked with a single usb camera (so without the switch), and this works as it should, so the switch is definitely the culprit

JarlLemmens, 2023-09-25:
@Erol444 I managed to just load a png image I stored earlier as input for the estimate pose function, which is giving me the matrices. Only when interpreting the results I am still getting an offset of +- 10 cm.
So I have an object point in pixel space with a measured depth (aruco marker) for each of the cameras.
I want to be able to transform these based on the extrinsic information so I would get the same x,y,z information from both cameras. My approach was to convert these (u,v,z) points to (X,Y,Z) space using the individual camera intrinsics, and then using the cam_to_world matrices to move both points to the same origin. But this gives the large offset. Would there be something going wrong here?

Erol444, 2023-09-25:
Hi @JarlLemmens , Not sure, this seems like opencv issue that it didn't estimate the pose correctly, perhaps high-res images provide better accuracy?

JarlLemmens, 2023-09-25:
@Erol444 but the approach to go from image coordinates to camera space (by using inverse of intrinsic matrix) and only then use the cam_to_world matrix to go to world coordinates, if I do this for both cameras, they should end up to be the same right? or at least very close

Erol444, 2023-09-25:
@JarlLemmens  Yes, that's the idea. If you are fusing pointclodus, then you'd also want to use pointcliud registration (eg. ICP) afterwards, so it's more accurate.