luxonis/depthai #784: [Support] Wrong input size when converting EfficientDet-lite3

abroekhof, 2022-09-02:
Doing the following to convert EfficientDet-lite3 from https://github.com/google/automl/tree/master/efficientdet:
```
$ python3 model_inspect.py --runmode=saved_model --model_name=efficientdet-lite3  --ckpt_path=efficientdet-lite3 --saved_model_dir=saved_model/modeldir

$ mo --reverse_input_channels --input_model ../efficientdet/efficientdet-lite3/saved_model/efficientdet-lite3_frozen.pb --transformations_config openvino_env/lib/python3.9/site-packages/openvino/tools/mo/front/tf/automl_efficientdet.json --input image_arrays --tensorboard_logdir . --input_shape [1,512,512,3]
```
Using the `blobconverter` webapp to convert the OpenVINO model, tried with both the defaults settings and setting `-iml NHWC -il NHWC`.

Always the same error when running on OAK-D PRO:
```
[mxid] [20.1] [168.777] [NeuralNetwork(1)] [warning] Input image (512x512) does not match NN (3x512)
```

Unclear why this is different from the EfficientDet in the OMZ, which I based this conversion on. What am I doing wrong?

Erol444, 2022-09-02:
Responding on discord: https://discord.com/channels/790680891252932659/1015080793455997000/1015185126386311168

abroekhof, 2022-09-02:
Was able to get this working with the following:

```
mo --reverse_input_channels --input_model ../efficientdet/efficientdet-lite3/saved_model/efficientdet-lite3_frozen.pb --transformations_config openvino_env/lib/python3.9/site-packages/openvino/tools/mo/front/tf/automl_efficientdet.json --input image_arrays --tensorboard_logdir . --input_shape [1,512,512,3] --layout "NHWC->NCHW" --data_type=FP16
```

I used the default settings for `blobconverter` in OpenVINO mode. 

I guess the NN node in DepthAI expects CHW layout?

closed 2022-09-02 by abroekhof

Erol444, 2022-09-03:
Hi @abroekhof , NN node doesn't expect anythin - only the model expects a specific layout. ColorCamera can output either planar or interleaved (`colorCam.setInterleaved(bool)`).
Thanks, Erik

abroekhof, 2022-09-07:
Maybe blobconverter expects CHW then? Might be worth checking in with whomever created these models: https://github.com/luxonis/blobconverter/blob/master/models/2022_1/efficientdet-d1-tf/model.yml

Would be good to add some documentation around how those are created and what the expected layout should be.

themarpe, 2022-09-07:
@abroekhof 

OpenVINO models usually (or always) default to CHW & BGR color order. When converting they highly recommend to convert to such input, such that feeding images into the model then only requires the knowledge of dimensions

abroekhof, 2022-09-07:
@themarpe, oh, wasn't aware of this recommendation—where do they say this? It's ambiguous since the XML of the IR pretty clearly specifies the input. If I leave the `--layout NHWC`, I get 
```
<?xml version="1.0" ?>
<net name="efficientdet-lite3_frozen" version="11">
        <layers>
                <layer id="0" name="image_arrays/placeholder_port_0" type="Parameter" version="opset1">
                        <data shape="1,512,512,3" element_type="f32"/>
                        <rt_info>
                                <attribute name="fused_names" version="0" value="image_arrays/placeholder_port_0"/>
                                <attribute name="old_api_map_element_type" version="0" value="f16"/>
                                <attribute name="old_api_map_order" version="0" value="0, 2, 3, 1"/>
                        </rt_info>
                        <output>
                                <port id="0" precision="FP32" names="0:image_arrays,image_arrays">
                                        <dim>1</dim>
                                        <dim>512</dim>
                                        <dim>512</dim>
                                        <dim>3</dim>
                                        <rt_info>
                                                <attribute name="layout" version="0" layout="[N,H,W,C]"/>
```

While if I set `--layout "NHWC->NCHW"`, I get:

```
<?xml version="1.0" ?>
<net name="efficientdet-lite3_frozen" version="11">
        <layers>
                <layer id="0" name="image_arrays/placeholder_port_0" type="Parameter" version="opset1">
                        <data shape="1,3,512,512" element_type="f32"/>
                        <rt_info>
                                <attribute name="fused_names" version="0" value="image_arrays/placeholder_port_0, image_arrays/placeholder_port_0/Transpose([0 3 1 2])"/>
                                <attribute name="old_api_map_element_type" version="0" value="f16"/>
                                <attribute name="old_api_map_order" version="0" value="0, 2, 3, 1"/>
                        </rt_info>
                        <output>
                                <port id="0" precision="FP32" names="0:image_arrays,Cast_1:0,image_arrays">
                                        <dim>1</dim>
                                        <dim>3</dim>
                                        <dim>512</dim>
                                        <dim>512</dim>
                                        <rt_info>
                                                <attribute name="layout" version="0" layout="[N,C,H,W]"/>
```

So it seems like `blobconverter` should be able to parse the layout and determine what the input to the NN is. It seems like there may be a bug there.

themarpe, 2022-09-12:
CC: @tersekmatija 

The conversion recommendation is stated here: https://docs.openvino.ai/2020.2/_docs_MO_DG_prepare_model_convert_model_Converting_Model_General.html#when_to_reverse_input_channels

But it also depends on the actual dataset at use, etc...

tersekmatija, 2022-09-12:
So blobconverter works on either NCHW or NHWC, in fact it is using the compile_tool from OpenVINO in the background, so anything that's possible to compile with that compiler should be possible to compile with blobconverter. And you've managed to compile the blob as you've said, but the error was 
`[mxid] [20.1] [168.777] [NeuralNetwork(1)] [warning] Input image (512x512) does not match NN (3x512)`

This means that while your model was expecting NHWC, it likely received NCHW images or vice versa. As Erik said, setting `colorCam.setInterleaved(bool)` should work to my knowledge. Have you tried that?