luxonis/depthai #788: [BUG] exited with status 100 at line 118

Weber-AILand, 2022-09-08:
I don't know how to better describe it than the screenshot already shows. 
Using the OAK-D-POE for the first time, first time set up. 

I am Working in Ubuntu 20.04 atm, have a virtual environment set up for it with python 3.10. Furthermore, I can provide additional information, not sure which would be useful to inspect this problem further.

Cheers

[image]

brmarkus, 2022-09-08:
Can you describe your environment in more details - what HW are you running on? 
How have you set-up your environment, e.g. creating the virtual-env?

Which exact Ubuntu OS do you use, is it a public distribution or your own? Is it a stock-ubuntu?
To circumvent local OS-config issues (e.g. like the multiple repo entries) - would it be possible to use a "clean" Ubuntu docker container, start it with host-network and proxy ENV variables - and call the install-script?

Are you behind a corporate network proxy? Can you check whether the failing repo-URLs can be resolved in your environment, e.g. via a web-browser or a wget?

Weber-AILand, 2022-09-13:
Hello Markus,

Thanks for the immediate response. Attached you find my try to immediately install without going trough a virt env. Meanwhile my colleague was able to immediately get it running on his machine.

what HW are you running on?
I have the G513QR from Asus so basically https://rog.asus.com/de/laptops/rog-strix/2021-rog-strix-g15-series/spec/
Relevant (I guess) RTX™ 3070 Laptop GPU
AMD Ryzen 9 5000 Series

creating the virtual-env?
Yes, the first attempt was set up with a virtual environment with anaconda. At creation time I installed python3 and nothing else. Then I tried to follow the instructions.

Which exact Ubuntu OS do you use, is it a public distribution or your own? Is it a stock-ubuntu?
Using of the rack Ubuntu 20.04 focal (atm 20.04.3 LTS)

Are you behind a corporate network proxy?
No

Can you check whether the failing repo-URLs can be resolved in your environment, e.g. via a web-browser or a wget?
Yes and yes for http://ppa.launchpad.net/webupd8team/atom/ubuntu/ with regards to the missing IP I found https://stackoverflow.com/questions/72284245/apt-upgrade-failing-keys but that doesn’t tell me anything as I am still a light Linux user.

To circumvent local OS-config issues (e.g. like the multiple repo entries) - would it be possible to use a "clean" Ubuntu docker container, start it with host-network and proxy ENV variables - and call the install-script?
Having trouble with docker atm, I will look into a clean reinstall, how about a clean ubuntu in a VM with virtual box? Unless that creates known issues

Hope this helps with the search for a solution. If you need more/other information feel free to ask.

Cheers,
Malte Weber

Weber-AILand, 2022-09-13:
[image]

The aforementioned screenshot

themarpe, 2022-09-14:
@Weber-AILand 

Usually `apt-get update` does not error out on such cases, but it seems that the `nvidia` & `atom` repo sources are throwing it off - an solution to that is to remove/fix those two particular sources and retry.

Another option is to downloading the script (`install_dependencies.sh`), removing the 3rd line (`set -x`) and then launch it: `bash install_dependencies.sh` and check if it works as expected for you.