luxonis/depthai #790: [BUG] {win 10 - RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND}

TheBricktop, 2022-09-14:
**Describe the bug**
Fresh install of python 3.8.5 virtualenv installed depthai, numpy, opencv-contrib running gives me single lens click in oakd-1 and then it throws aforementioned error

**Minimal Reproducible Example**
pip install the depthai lib and opencv and run any demo.

**Expected behavior**
Demo run properly like in the video.

**Attach system log**

        (venv) PS B:\Github\depthAI_SKAKUN> & b:/Github/depthAI_SKAKUN/venv/Scripts/python.exe b:/Github/depthAI_SKAKUN/stereoview_BW.py
        Traceback (most recent call last):
          File "b:/Github/depthAI_SKAKUN/stereoview_BW.py", line 30, in <module>
            with dai.Device(pipeline) as device:
        RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

brmarkus, 2022-09-14:
Can you provide more details about how you have set-up your environment, including the HOST, please?

Because the installation requires a UDEF-rule file to be added (because the contained MyriadX-chip changes from HIGHSPEED to SUPERSPEED while loading and starting the firmware, thus UDEF needs to detect the MyriadX device to re-appear as a different device (again), but under a different VENDORID-DEVICEID.

TheBricktop, 2022-09-14:
I was using the official Win10 install guide from depthai docs site and it is not mentioning nothing about UDEF under the win10 installation
How should I do this?

brmarkus, 2022-09-14:
You are right... under MS-Win UDEF doesn't make sense, this is for Linux environments only.

Do you use the "original USB3 cables", are you sure they are USB3-capable? Do you have an actively powered USB-hub inbetween? Are you sure you connected the USB3-cable into a USB3-capable port on your machine?

TheBricktop, 2022-09-14:
Yeah this is the original usb3.1 cable that came with the oakd1 but it seem that it stopped working, and im positively sure that it worked with the same port as before when it was working.
EDIT : changed cable and it is working again
EDIT: ive tried this with godot-python extension and it tells me that there is no device attached to the pc
so ive came back and tried the virtualenv that was working like 5 minutes before and it shows the same error
what gives?

brmarkus, 2022-09-14:
Could there be a loose connection?
Could there be an instable power-supply on your machine?
Would you have an USB3-capable actively powered USB-hub, using USB3-cables between computer and hub and between hub and the camera?

TheBricktop, 2022-09-15:
well now,
it is sometwhat obscure but when the godot-python python server starts it initialises for some reason some generic depth ai pipeline effectively blocking the device from being available to another call thus device is unavailable.

brmarkus, 2022-09-15:
Do I understand you right that in your environment there are multiple independent processes actively accessing the OAK-D device concurrently?

taaneredsov, 2022-09-15:
Hello, I faced the same problem. I used USBDeview (https://www.nirsoft.net/utils/usb_devices_view.html) to uninstall/delete the Movidius MyriadX entry. After reconnecting the camera reinstalled and all worked again.