luxonis/depthai #798: How to convert the pixel coordinates from mono to RGB?

re-this, 2022-09-23:
Hello, 

I ran some algorithms and found the location of the object in **mono camera**.
Now, what I want to do is visualizing something like bounding box of that object in **RGB camera**.
But I don't know how to convert coordinates between them.

Please let me know the way

Thank you

Erol444, 2022-09-24:
Hi @re-this ,
That would be quite complicated, and depthai currently doesn't support that. You can, however, aling depth to eg. RGB camera ([example here](https://docs.luxonis.com/projects/api/en/latest/samples/StereoDepth/rgb_depth_aligned/#rgb-depth-alignment)).

You could do this on the host computer though - you'd need to use camera matrix (see [calibration reader example](https://docs.luxonis.com/projects/api/en/latest/samples/calibration/calibration_reader/#calibration-reader)) and current depth values in order to map bounding box from mono cam to rgb cam. Thoughts?
thanks, Erik

re-this, 2022-09-26:
Hi, @Erol444 

As you commented, I converted pixel coordinates of mono camera into rgb, using intrinsics, extrinsics matrices and depth.
But, the accuracy of point is quite low.
Maybe, it's due to the low quality of depth.

Anyway, really thank you!

closed 2022-09-26 by re-this