luxonis/depthai #807: [BUG] {Error running python3 depthai_demo.py on Ubuntu 22.04.1 LTS}

czernalex, 2022-10-04:
Hi,

I wanted to test depthai demo app on OAK-D Lite camera, but I ran into following problems. I successfully ran these commands from my terminal:
```
git clone https://github.com/luxonis/depthai.git
cd depthai
python3 install_requirements.py
```

But when I tried to run `python3 depthai_demo.py` I received this error message:

```
Depthai version installed:  2.17.3.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
```

Then I visited [Installation page](https://docs.luxonis.com/projects/api/en/latest/install/) for additional OS-specific instructions and since I am working on Ubuntu, I tried to run `sudo wget -qO- https://docs.luxonis.com/install_dependencies.sh | bash`, but got this error:

```
Hit:1 http://cz.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://cz.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:3 http://cz.archive.ubuntu.com/ubuntu jammy-backports InRelease
Get:4 http://security.ubuntu.com/ubuntu jammy-security InRelease [110 kB]
Get:5 http://security.ubuntu.com/ubuntu jammy-security/main i386 Packages [136 kB]
Get:6 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [352 kB]
Get:7 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [287 kB]
Fetched 886 kB in 1s (1 261 kB/s)                          
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libdc1394-22-dev

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 119
```

After some [googling](https://askubuntu.com/questions/1407580/unable-to-locate-package-libdc1394-22-dev) I tried to manually edit the `install_dependencies.sh` file and changed line 26 from `libdc1394-22-dev` to `libdc1394-dev` and since this moment everything works as expected.

Erol444, 2022-10-06:
Hi @czernalex ,
Thank you for the detailed bug report and the solution to fix this! We will edit the install_dependencies.sh asap!
Kind regards, Erik

themarpe, 2022-10-19:
Closed by: https://github.com/luxonis/depthai-python/commit/8ede14b10ff4c9b455681d1e92b184c42090f1c3

closed 2022-10-19 by themarpe