luxonis/depthai #812: Temperature above 105C. Device will shutdown if not reduced

ddwy, 2022-10-17:
When I execute the following command, the following error occurs:

python3 depthai_demo.py 

[1844301011EB111300] [1.2] [79.648] [system] [critical] Temperature above 105C. Device will shutdown if not reduced. Note: If this is unexpected please reach out to support@luxonis.com to discuss/debug.
[CRITICAL] [0:01:19.648111] Temperature above 105C. Device will shutdown if not reduced. Note: If this is unexpected please reach out to support@luxonis.com to discuss/debug.
Chip temperature - average: 104.39, css: 105.29, mss: 103.99, upa0: 105.62, upa1: 102.67

File "depthai_demo.py", line 655, in run
    self.instance.run_all(self.conf)
  File "depthai_demo.py", line 132, in run_all
    self.run()
  File "depthai_demo.py", line 316, in run
    self.loop()
  File "depthai_demo.py", line 422, in loop
    logs = self._logOut.tryGetAll()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'systemLogger' (X_LINK_ERROR)'

Erol444, 2022-10-17:
Hi @ddwy , what device are you using? If it's OAK-D-Lite, what version is it (FF/AF)? How long after the start does this problem show up?
Thanks, Erik

ddwy, 2022-10-17:
> Hi @ddwy , what device are you using? If it's OAK-D-Lite, what version is it (FF/AF)? How long after the start does this problem show up? Thanks, Erik

I am using OAK-D-Lite.When I use this camera for a few minutes, the above error will occur.

Erol444, 2022-10-17:
Hi @ddwy ,
We apologize for the inconvenience. Please send an email to support@luxonis.com with the link to this Issue, we will take care of it.
Thanks, Erik

ddwy, 2022-10-17:
> Hi @ddwy , We apologize for the inconvenience. Please send an email to [support@luxonis.com](mailto:support@luxonis.com) with the link to this Issue, we will take care of it. Thanks, Erik

I've already sent an email.

ecmnet, 2022-10-19:
Got any answer? Same issue here with OAK-D Lite

Erol444, 2022-10-19:
@ecmnet we apologize for the issue, please send an email to support@luxonis.com, we will take care of it. 
Thanks, Erik

Erol444, 2023-02-06:
For anyone else having similar problem, please contact support@luxonis.com. Closing now due to inactivity.

closed 2023-02-06 by Erol444