luxonis/depthai #828: [BUG] {Title of the bug}

JimXu1989, 2022-11-02:
Hello,
I have a OAK-D-PoE, when I use the depthai_demo.py, I got the following error:

xss@xss-AERO-15-X9:~/SoftWare/depthai$ python3 depthai_demo.py
Using depthai module from:  /home/xss/.local/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
Depthai version installed:  2.18.0.0
Setting up demo...
Available devices:
[0] 14442C10510034D700 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
[14442C10510034D700] [169.254.1.222] [24.468] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
[CRITICAL] [0:00:24.468541] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
How should I fix it?
Thanks

Erol444, 2022-11-02:
Hi @JimXu1989 ,
Could you go through the [troubleshooting docs](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html) of OAK POE cameras? 
Thanks, Erik

iam-vishnu, 2022-11-09:
Hi @Erol444 , Thanks. Even I got the same error with OAK-D-POE. As per your suggestion, installed 2.17.4, and it seems working, It didn't crashed.

### Error

```
Depthai version installed:  2.18.0.0
Setting up demo...
Available devices:
[0] 18443010F1EFFB0F00 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
[18443010F1EFFB0F00] [192.168.41.87] [259.250] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
[CRITICAL] [0:04:19.250872] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
Stopping demo...
```
## After fix, 
```
Depthai version installed:  2.17.4.0
Setting up demo...
Available devices:
[0] 18443010F1EFFB0F00 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
```

iam-shanmukha, 2022-11-09:
Hi @Erol444 , Thanks. Even I got the same error with OAK-D-POE. As per your suggestion, installed 2.17.4, and it seems working, It didn't crashed.

### Error

```
Depthai version installed:  2.18.0.0
Setting up demo...
Available devices:
[0] 18443010F1EFFB0F00 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
[18443010F1EFFB0F00] [192.168.41.87] [259.250] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
[CRITICAL] [0:04:19.250872] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 09, address: 801F6FC8' '0'
Stopping demo...
```
## After fix, 
```
Depthai version installed:  2.17.4.0
Setting up demo...
Available devices:
[0] 18443010F1EFFB0F00 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...
```

Erol444, 2022-11-09:
Hi @iam-shanmukha , that's great to hear! Feel free to close this Issue if works as expected now.
Thanks, Erik

alex-luxonis, 2022-11-09:
The requirements are updated to install `2.19.0.0`, could you pull the latest.

iam-shanmukha, 2022-11-09:
Hi @alex-luxonis , Yeah, 2.19.0.0 also working.
```
Depthai version installed:  2.19.0.0
Setting up demo...
Available devices:
[0] 18443010F1EFFB0F00 [X_LINK_BOOTLOADER]
Enabling low-bandwidth mode due to connection mode... (protocol: XLinkProtocol.X_LINK_TCP_IP)
Setting PoE video quality to 50 to reduce latency...
Creating MJPEG link for MonoCamera node and left xlink stream...
Creating MJPEG link for MonoCamera node and right xlink stream...
Creating MJPEG link for ColorCamera node and color xlink stream...

```

closed 2022-11-21 by themarpe