luxonis/depthai #833: Unable to locate package libdc1394-dev

RAO-3D, 2022-11-05:
Similar to earlier issue ( https://github.com/luxonis/depthai/issues/807#issue-1396735834) 

I am using earlier version of Ubuntu (18.04 Bionic) and get:

_Reading package lists... Done
Building dependency tree      
Reading state information... Done
E: Unable to locate package libdc1394-dev_

_Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 100 at line 132_

Must I update to Ubuntu 21.10 ?

themarpe, 2022-11-08:
@RAO-3D we support Ubuntu 20.04 and Ubuntu 22.04

That'd be the simplest to get up and running.

Otherwise you can install the udev rules manually and use the examples and library over at `depthai-python` as is.

closed 2022-11-08 by themarpe

RAO-3D, 2022-11-08:
Thanks - Updated and got things working.