luxonis/depthai #842: [BUG] {Wrong intrinsic matrix for resized RGB image}

AntonioConsiglio, 2022-11-10:
**Describe the bug**

Wrong intrinsic matrix for resized RGB image. While in version 2.15.0.0 the intrinsic matrix is correct, in version depthai >= 2.17.3.1 the intrinisc matrix is different in that case and it's wrong.

**Minimal Reproducible Example**

`self.device_ = dhai.Device(pipeline,usb2Mode=True)`
`calibration_info = self.device_.readCalibration()`
`intrinsic_matrix = calibration_info.getCameraIntrinsics(dhai.CameraBoardSocket.RGB,resizeHeight=480,resizeWidth=640)`

**Additional context**
camera = OAK-D Lite
depthai version tested: 2.15.0.0 (work) - 2.17.3.1 (wrong) - 2.19.0.0 (wrong)

brmarkus, 2022-11-11:
Can you provide sample output of a wrong and a correct version, please?

AntonioConsiglio, 2022-11-11:
**right output**
[image]

**wrong output**
[image]

daxoft, 2022-11-14:
Hi @AntonioConsiglio sorry for the issue, can you share the output of [calibration_dump](https://github.com/luxonis/depthai-python/blob/main/examples/calibration/calibration_dump.py) script? And also check if it's the same for both depthai versions?

AntonioConsiglio, 2022-11-14:
Hello @daxoft, 
[calibration_dump.zip](https://github.com/luxonis/depthai/files/10005454/calibration_dump.zip)
Here you can find the json file made using the calibration_dump script. I've checked the user_calibration*.json and the output is the same for both 1.17.0.0 and 1.19.1.0 version.

Looking at the source code, I saw that the method used to request calibration information has changed. A flag was added and also the "calculation" part was changed. For a resize with the same ratio the calculation of the new intrinsic matrix is quite obvious, probably because my ratio does not remain the same as the original this leads to a calculation error.

Thanks.