luxonis/depthai #852: [BUG] OAKD-POE camera pipeline data acquisition fails after approx. 100 frames when setLeftRightCheck is set to True

kartik-innovasea, 2022-11-22:
Hi, 

I am trying to create a pipeline on the OAKD-PoE camera to parse synced image and depth frames. I see that `setLeftRightCheck` is the LR check corresponding to the LR [consistency check as posted here in the luxonis documentation](https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/).
The code is similar to this [depthai-experiments script ](https://github.com/luxonis/depthai-experiments/blob/master/gen2-human-machine-safety/main.py)

`   print("Creating pipeline...")
    pipeline = dai.Pipeline()

    cam = pipeline.create(dai.node.ColorCamera)
    cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
    cam.setIspScale(2, 3) # To match 720P mono cameras
    cam.setBoardSocket(dai.CameraBoardSocket.RGB)
    cam.initialControl.setManualFocus(140)
    cam.Fps(30)
    # For MobileNet NN
    cam.setColorOrder(dai.ColorCameraProperties.ColorOrder.BGR)
    cam.setPreviewSize(300, 300)
    cam.setInterleaved(False)

    isp_xout = pipeline.create(dai.node.XLinkOut)
    isp_xout.setStreamName("cam")
    cam.isp.link(isp_xout.input)

    # Creating left/right mono cameras for StereoDepth
    left = pipeline.create(dai.node.MonoCamera)
    left.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    left.setBoardSocket(dai.CameraBoardSocket.LEFT)
    left.Fps(30)
    right = pipeline.create(dai.node.MonoCamera)
    right.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    right.setBoardSocket(dai.CameraBoardSocket.RIGHT)
    right.Fps(30)
    # Create StereoDepth node that will produce the depth map
    stereo = pipeline.create(dai.node.StereoDepth)
    stereo.initialConfig.setConfidenceThreshold(245)
    stereo.initialConfig.setMedianFilter(dai.StereoDepthProperties.MedianFilter.KERNEL_7x7)
    stereo.setLeftRightCheck(True) ### ****************ISSUE WITH THIS API CALL********************
    stereo.setDepthAlign(dai.CameraBoardSocket.RGB)
    left.out.link(stereo.left)
    right.out.link(stereo.right)

    sdn = pipeline.create(dai.node.MobileNetSpatialDetectionNetwork)
    sdn.setBlobPath(blobconverter.from_zoo(name="mobilenet-ssd", shaves=6))
    sdn.setConfidenceThreshold(0.5)
    sdn.input.setBlocking(False)
    sdn.setBoundingBoxScaleFactor(0.2)
    sdn.setDepthLowerThreshold(DEPTH_THRESH_LOW)
    sdn.setDepthUpperThreshold(DEPTH_THRESH_HIGH)

    cam.preview.link(sdn.input)
    stereo.depth.link(sdn.inputDepth)

    sdn_out = pipeline.create(dai.node.XLinkOut)
    sdn_out.setStreamName("det")
    sdn.out.link(sdn_out.input)

    depth_out = pipeline.create(dai.node.XLinkOut)
    depth_out.setStreamName("depth")
    sdn.passthroughDepth.link(depth_out.input)

    print("Pipeline created.")

    with dai.Device(pipeline, device_info) as device:

        vidQ = device.getOutputQueue(name="cam", maxSize=1, blocking=False)
        depthQ = device.getOutputQueue(name="depth", maxSize=1, blocking=False)

        depthFrame = None
        frame = None
        cnt = 0
        while True:
            in_rgb = vidQ.tryGet()
            rgbFrame = in_rgb.getCvFrame()

            cv2.imshow("frame", rgbFrame)
            cv2.waitKey(30)

`

szabi-luxonis, 2022-11-22:
@kartik-innovasea it's not possible that enabling left-right check causes the problem, because it's enabled by default.
`stereo.setLeftRightCheck(True)` does nothing, since it's already `True` by default.

kartikmadhira1, 2022-11-22:
> @kartik-innovasea it's not possible that enabling left-right check causes the problem, because it's enabled by default. `stereo.setLeftRightCheck(True)` does nothing, since it's already `True` by default.

Thanks for your reply. What could be the possible issue then? I am trying to save rgb and depth frames but the pipeline freezes after 70-80 frames. I am using an OAKD-PoE with 50 ft cable.

szabi-luxonis, 2022-11-22:
The code you posted in the issue doesn't work @kartik-innovasea .
Here is the corrected one, that runs:

```
import depthai as dai
import cv2
import blobconverter

pipeline = dai.Pipeline()
cam = pipeline.create(dai.node.ColorCamera)
cam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
cam.setIspScale(2, 3) # To match 720P mono cameras
cam.setBoardSocket(dai.CameraBoardSocket.RGB)
cam.initialControl.setManualFocus(140)
cam.setFps(30)
# For MobileNet NN
cam.setColorOrder(dai.ColorCameraProperties.ColorOrder.BGR)
cam.setPreviewSize(300, 300)
cam.setInterleaved(False)

isp_xout = pipeline.create(dai.node.XLinkOut)
isp_xout.setStreamName("cam")
cam.isp.link(isp_xout.input)

# Creating left/right mono cameras for StereoDepth
left = pipeline.create(dai.node.MonoCamera)
left.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
left.setBoardSocket(dai.CameraBoardSocket.LEFT)
left.setFps(30)
right = pipeline.create(dai.node.MonoCamera)
right.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
right.setBoardSocket(dai.CameraBoardSocket.RIGHT)
right.setFps(30)
# Create StereoDepth node that will produce the depth map
stereo = pipeline.create(dai.node.StereoDepth)
stereo.initialConfig.setConfidenceThreshold(245)
stereo.initialConfig.setMedianFilter(dai.StereoDepthProperties.MedianFilter.KERNEL_7x7)
stereo.setLeftRightCheck(True) ### ****************ISSUE WITH THIS API CALL********************
stereo.setDepthAlign(dai.CameraBoardSocket.RGB)
left.out.link(stereo.left)
right.out.link(stereo.right)

sdn = pipeline.create(dai.node.MobileNetSpatialDetectionNetwork)
sdn.setBlobPath(blobconverter.from_zoo(name="mobilenet-ssd", shaves=6))
sdn.setConfidenceThreshold(0.5)
sdn.input.setBlocking(False)
sdn.setBoundingBoxScaleFactor(0.2)
# sdn.setDepthLowerThreshold(DEPTH_THRESH_LOW)
# sdn.setDepthUpperThreshold(DEPTH_THRESH_HIGH)

cam.preview.link(sdn.input)
stereo.depth.link(sdn.inputDepth)

sdn_out = pipeline.create(dai.node.XLinkOut)
sdn_out.setStreamName("det")
sdn.out.link(sdn_out.input)

depth_out = pipeline.create(dai.node.XLinkOut)
depth_out.setStreamName("depth")
sdn.passthroughDepth.link(depth_out.input)

print("Pipeline created.")

with dai.Device(pipeline) as device:

    vidQ = device.getOutputQueue(name="cam", maxSize=1, blocking=False)
    depthQ = device.getOutputQueue(name="depth", maxSize=1, blocking=False)

    depthFrame = None
    frame = None
    cnt = 0
    while True:
        in_rgb = vidQ.get()
        rgbFrame = in_rgb.getCvFrame()

        cv2.imshow("frame", rgbFrame)
        cv2.waitKey(30)
```

kartik-innovasea, 2022-11-22:
> @kartik-innovasea it's not possible that enabling left-right check causes the problem, because it's enabled by default. `stereo.setLeftRightCheck(True)` does nothing, since it's already `True` by default.

Thank you for your reply. What could be the possible issue with the above minimal snippet then? The frame freezes after few 100 frames? I see that decreasing the number of shaves on the spatial network blob for `MobileNetSpatialDetectionNetwork` increase the amount of frames acquired before freezing. I am using OAKD PoE with 50 ft cable.

kartik-innovasea, 2022-11-22:
@szabi-luxonis Thanks for pointing that out and apologies for posting an incorrect minimal snippet of the code. I would like to point out that this [human-machine-safety](https://github.com/luxonis/depthai-experiments/blob/master/gen2-human-machine-safety/main.py) script also uses the` sdn_out `dequeing. But with this script too I am able to get only a few 1000 frames. It does freeze, albeit after like 10 minutes. Only addition to the code was that I was not cropping the frame as per line 307 of the depthai-experiments snippet I posted.

kartik-innovasea, 2022-11-29:
Thank you @szabi-luxonis. This actually helped. I think the FPS being set to 30 fixes the issue of frames freezing.

closed 2022-11-29 by kartik-innovasea