luxonis/depthai #858: [BUG] Could not reconstruct & visualize stereo depth from recording of left/right

martin0258, 2022-12-02:
**Check if issue already exists**

> 
> - Google it (e.g. *error xy* github luxonis depthai)
> - Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

Yes, but I found nothing related

**Describe the bug**
As title, I following the usage of [replaying](https://docs.luxonis.com/projects/sdk/en/latest/features/replaying/) from docs, but I could not reconstruct & visualize stereo depth from the recording of the left/right monoframes.

**Minimal Reproducible Example**
By using the recording [18443010B1C8361300.zip](https://github.com/luxonis/depthai/files/10144079/18443010B1C8361300.zip) and run the following code

```python
from depthai_sdk import OakCamera

with OakCamera(replay='18443010B1C8361300') as oak:
    color = oak.create_camera('color')
    left = oak.create_camera('left')
    right = oak.create_camera('right')
    stereo = oak.create_stereo(left=left, right=right)
    oak.visualize([color, left, right, stereo.out.depth], scale=1/3)
    oak.start(blocking=True)
```

I got the following runtime error
```
Available streams from recording: ['color', 'right', 'left']
Closing OAK camera
Closing replay
Traceback (most recent call last):
  File "/home/martin/Projects/sock_rotation_ai_poc/replay.py", line 10, in <module>
    oak.start(blocking=True)
  File "/home/martin/Projects/sock_rotation_ai_poc/depthai/depthai_sdk/src/depthai_sdk/oak_camera.py", line 267, in start
    self.build()  # Build the pipeline
  File "/home/martin/Projects/sock_rotation_ai_poc/depthai/depthai_sdk/src/depthai_sdk/oak_camera.py", line 354, in build
    xouts = out.setup(self._pipeline, self._oak.device, names)
  File "/home/martin/Projects/sock_rotation_ai_poc/depthai/depthai_sdk/src/depthai_sdk/classes/output_config.py", line 53, in setup
    xoutbase: XoutBase = self.output(pipeline, device)
  File "/home/martin/Projects/sock_rotation_ai_poc/depthai/depthai_sdk/src/depthai_sdk/components/stereo_component.py", line 214, in depth
    mono_frames=StreamXout(self._comp.node.id, self._comp.right.out),
  File "/home/martin/Projects/sock_rotation_ai_poc/depthai/depthai_sdk/src/depthai_sdk/oak_outputs/xout_base.py", line 17, in __init__
    self.name = f"{str(id)}_{out.name}"
AttributeError: 'Out' object has no attribute 'name'
```

**Expected behavior**
It should reconstruct & visualize the depth.

**Screenshots**
NA

**Pipeline Graph**

> Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).

I guess the graph is not so helpful in this case? The MRE is quite simple.

**Attach system log**
```json
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.0-56-generic-x86_64-with-glibc2.31",
    "processor": "x86_64",
    "python_build": "main Nov 22 2022 15:55:03",
    "python_compiler": "GCC 10.4.0",
    "python_implementation": "CPython",
    "python_version": "3.9.15",
[... 199 lines cut ...]
        "websockets==10.4",
        "Werkzeug==2.2.2",
        "wheel==0.38.4",
        "wrapt==1.14.1",
        "wurlitzer==3.0.3",
        "xmltodict==0.13.0",
        "yarl==1.8.1",
        "zipp==3.11.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux SS-KINGS 5.15.0-56-generic #62~20.04.1-Ubuntu SMP Tue Nov 22 21:24:20 UTC 2022 x86_64 x86_64"
    ]
}
```

**Additional context**
Forgive me if it's my incorrect API usage and not a bug.

I simply wrote the code because the docs said so. Maybe my interpretation of the docs is incorrect?
> Because [Recording](https://docs.luxonis.com/projects/sdk/en/latest/features/recording/#recording) saves calibration data and can save synchronized left+right mono streames we can achieve full depth reconstruction.

If I changed `xout_base.py` to bypass the error as a workaround, there was no error but depth was not shown -- only `color`, `left, `right` were visualized as 3 frames, respectively.

Erol444, 2022-12-03:
Hi @martin0258 , could you post the graph as well? You can create it via `oak.show_graph()`

martin0258, 2022-12-03:
@Erol444 
[image]

daniilpastukhov, 2022-12-03:
Hi @martin0258, thanks for the issue. I've found the bug that causes tha error. When using replay, `depthai_sdk.CameraComponent` is being used instead of expected `depthai.MonoCamera` for left/right videos. I am working on a fix, I will let you now once it's fixed.

closed 2022-12-04 by daniilpastukhov