luxonis/depthai #867: [Question] Demo on M1 Mac Using Oak D Lite doesn't show options

gomako, 2022-12-08:
When running `depthai_demo.py` on my M1 macbook pro using Oak D Lite, I get a color and depth preview window but I don't get the options to change models etc. 

I am using Conda to get OpenCV running on my machine as outlined here https://blog.roboflow.com/m1-opencv/ and it's working well. I could never get it working on my older mac.

I pulled this repo, switched to my openCV conda environment, ran `python install_requirements.py` and then `python depthai_demo.py`

**Screenshots**

Expected:

[image]

Actual:
[image]

Not really sure what to do to get the full demo working?

Erol444, 2022-12-08:
Hi @gomako ,
By default it will use CV GUI on MacM1. You can force qt with `python3 depthai_demo.py -cv qt`. Let us know if that helps.
Thanks, Erik

gomako, 2022-12-08:
Hi @Erol444 Thanks that seemed to do the trick.

closed 2022-12-08 by gomako

mngyuan, 2023-05-30:
The above fix no longer works. I've gotten around it by doing the following

```
export PYTHONPATH=/opt/homebrew/lib/python3.11/site-packages:
python3 launcher/launcher.py
```

by copying the `install_depthai.sh` script's workaround for QT. The equivalent command `python3 depthai_demo.py -gt qt` complains about the QT backend not being available, but `pip install pyqt5` results in the following error in the `venv` on M1:

```
(venv) ~/Downloads  ➜  pip install pyqt5
Collecting pyqt5
  Downloading PyQt5-5.15.9.tar.gz (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 14.0 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Traceback (most recent call last):
        File "/Users/liuyue/Luxonis/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/Users/liuyue/Luxonis/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/liuyue/Luxonis/venv/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 152, in prepare_metadata_for_build_wheel
          whl_basename = backend.build_wheel(metadata_directory, config_settings)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/sipbuild/api.py", line 46, in build_wheel
          project = AbstractProject.bootstrap('wheel',
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/sipbuild/abstract_project.py", line 87, in bootstrap
          project.setup(pyproject, tool, tool_description)
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 586, in setup
          self.apply_user_defaults(tool)
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-install-ub1ae_et/pyqt5_4db4529b7277449098b4753ffc1e82c8/project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/pyqtbuild/project.py", line 70, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/sipbuild/project.py", line 237, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        File "/private/var/folders/9c/3h58n31d0dvc_fflxgtwftmr0000gq/T/pip-build-env-7t_fwxr0/overlay/lib/python3.11/site-packages/pyqtbuild/builder.py", line 69, in apply_user_defaults
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Will I just have to use my global python install on my M1 mac/should I use conda instead?

themarpe, 2023-05-30:
@mngyuan seems like pyqt5 isn't available for M1 - perhaps brew or some other package manager has a compiled wheel available for M1.