luxonis/depthai #874: [BUG] {[19443010E1C2F51200] [1.4] [58.518] [system] [critical] Fatal error. Please report to developers. Log: 'ResourceLocker' '358'}

Undertaker7533967, 2022-12-12:
**Check if issue already exists**
==>my issue is already exists while google it on github luxonis depthai.

**Describe the bug**
==> we are runing [depthai_hand_tracker] models by using visual studio platform.
when we are detecting our single hand very closed within 40cm. it will shut down immedeately and  no any detection. then we have to foced to clsoed  models .
we found out Fatal error , Please report to developers. Log: 'ResourceLocker' '358' 

**Minimal Reproducible Example**
Append the MRE to the bug report

(https://www.google.com/search?q=Fatal+error+%2C+Please+report+to+developers.+Log%3A+%27ResourceLocker%27+%27358%27+&ei=LJmWY5-IJ7uHr7wP6_KFqAQ&ved=0ahUKEwifzoWWi_P7AhW7w4sBHWt5AUUQ4dUDCA8&uact=5&oq=Fatal+error+%2C+Please+report+to+developers.+Log%3A+%27ResourceLocker%27+%27358%27+&gs_lcp=Cgxnd3Mtd2l6LXNlcnAQAzoKCAAQRxDWBBCwAzoECAAQQzoFCAAQgARKBAhBGABKBAhGGABQkQFY2WxgwndoAXABeACAATeIAV-SAQEymAEAoAEBoAECyAEKwAEB&sclient=gws-wiz-serp)](https://github.com/luxonis/depthai/issues/602)](https://github.com/luxonis/depthai/issues/602)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Pipeline Graph**
no

**Attach system log**
 - Provide output of [0803_hand_tracker_demo.py] with [hand_landmark_080_sh4.blob] and 
 -[hand_landmark_full_sh4.blob] and [hand_landmark_lite_sh4.blob ]
 -

**Additional context**==>[0803_hand_tracker_demo.py] as below source code

import HandTrackerRenderer, HandTrackerEdge
# import board
# import digitalio
import cv2
import tkinter as tk 
import tkinter.ttk as ttk
from PIL import Image, ImageTk, ImageFont, ImageDraw, Image
import time

# red_o = digitalio.DigitalInOut(board.C0)
# red_o.direction = digitalio.Direction.OUTPUT
# red_o.value = False

tracker_args = {"use_same_image" : True}

tracker = HandTrackerEdge.HandTracker(
input_src =  None,
use_lm = True,
use_world_landmarks = False,
use_gesture = True, 
xyz = False, 
solo = True,
crop = False,
resolution = "full", 
stats = True, 
trace = 0, 
use_handedness_average = True,
single_hand_tolerance_thresh = 5,
lm_nb_threads = 2, 
**tracker_args)

renderer = HandTrackerRenderer.HandTrackerRenderer(
    tracker = tracker,
    output = None)

# # GUI
# wd = tk.Tk() # 建立[根視窗]
# wd.title("Riko Machine Vision - HV-R100 - 手部安全防護") # 建立視窗名稱
# wd.iconbitmap("Riko.ico") # 建立視窗 icon 
# w = 1920
# h = 1080
# x = -10
# y = 0
# wd.geometry("%dx%d+%d+%d" %(w, h, x, y)) # 設定視窗大小及位置( + 左 & 上；- 右 & 下)
# pws = ttk.PanedWindow(wd, orient = tk.HORIZONTAL) # 全視窗父容器
# pws.pack( fill = tk.BOTH, expand = True) # expand 是否填滿視窗
# #======================================================================================
# #※ 左邊的畫面
# # 左邊的父容器
# pw = ttk.PanedWindow(pws, orient = tk.VERTICAL) 
# pws.add(pw, weight = 2)
# label_1 = tk.Label(pw, bg = "pale green", width = 1920, height = 1080) # 畫面影像容器
# pw.add(label_1, weight = 2)
#======================================================================================
# #※ 右邊的資訊列
# # 右邊的父容器
# pw2 = ttk.PanedWindow(pws, orient = tk.VERTICAL)
# pws.add(pw2, weight = 2)
# #-----------------------------------------------------------------------------------
# # 數據列容器
# bg_lf1 = "sky blue"
# labelframe_1 = tk.LabelFrame(pw2, bg = bg_lf1, width = 160, height = 700)
# pw2.add(labelframe_1, weight = 2)

# 組別
# min_d_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "手部安全防護 ", width = 10, 
#                          font = ("Times", 16, "bold"))
# min_d_label_1.grid(padx = 3, pady = 2, row = 0, column = 0, sticky =  tk.E + tk.W)

# min_d_label_4 = tk.Label(labelframe_1, bg = bg_lf1, width = 2, font = ("Times", 24, "bold"))
# min_d_label_4.grid(padx = 3, pady = 2, row = 1, column = 2, sticky =  tk.E + tk.W) 

# # 最近距離
# min_d_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "最近距離 : ", width = 8,
#                 font = ("Times", 24, "bold"))
# min_d_label_1.grid(padx = 3, pady = 2, row = 2, column = 0, sticky =  tk.E + tk.W)
# min_d_label_2 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_2.grid(padx = 3, pady = 2, row = 3, column = 0, sticky =  tk.E + tk.W)
# min_d_label_3 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_3.grid(padx = 3, pady = 2, row = 3, column = 1, sticky =  tk.E + tk.W) 
# min_d_label_4 = tk.Label(labelframe_1, bg = bg_lf1, width = 8, font = ("Times", 24, "bold"))
# min_d_label_4.grid(padx = 3, pady = 2, row = 4, column = 1, sticky =  tk.E + tk.W) 

# # fps
# fps_label_1 = tk.Label(labelframe_1, bg = bg_lf1, text = "FPS : ",  width = 8, 
#                        font = ("Times", 24, "bold"))
# fps_label_1.grid(padx = 3, pady = 2, row = 5, column = 0, sticky =  tk.W )
# fps_label_2 = tk.Label(labelframe_1, bg = bg_lf1, fg = "orange red", width = 8,
#                         font = ("Times", 24, "bold"))
# fps_label_2.grid(padx = 3, pady = 2, row = 5, column = 1, sticky =  tk.E + tk.W)

while True :
   
    
    i_number, frame, hands, bag = tracker.next_frame()
    print("i_number = ", i_number)
    
    if frame is None:
        break
    frame = renderer.draw(frame, hands, bag)
    
    # frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
    # imgtk = ImageTk.PhotoImage(image = Image.fromarray(frame))
    # label_1.imgtk = imgtk
    # label_1.config(image = imgtk)
    # label_1.update()
    # key = cv2.waitKey(1)
   

    key, frame = renderer.waitKey(delay = 1)
    cv2.imshow("Riko Machine Vision - HV-R100", frame)
    # frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
    # imgtk = ImageTk.PhotoImage(image = Image.fromarray(frame))
    # label_1.imgtk = imgtk
    # label_1.config(image = imgtk)
    # label_1.update()
 
    
    if key == 27 or key == ord('q') :
        break
    
renderer.exit()
tracker.exit()

Erol444, 2022-12-12:
Hi @Undertaker7533967 , I see you are using the handtracker demo. Since this wasn't developed by us, I would suggest opening an Issue on the original repo.
Thanks, Erik

Undertaker7533967, 2022-12-14:
Dear Erik:

Thank you for recommend, we have pass this issue to original developer repo.

all the best~~

closed 2022-12-14 by Erol444