luxonis/depthai #882: [BUG] OAK-D Pro W right mono camera distorted after calibration

ZumWoelkchen, 2022-12-22:
### **What:**

It appears that the calibration tool messes up calibration on OAK-D Pro W devices. I reproduced this behavior on two devices with the same result.

- The example images below show both left & right mono cameras pointing to the same grid.

[image]
[image]

### **How to reproduce:**

- Check the rectified right mono camera with factory calibration. Should be fine.
- Run calibration tool: python3 calibrate.py -s 2.45 -brd BW1098OBC -db
After successful calibration check the cameras again. Rectified left should look fine. Rectified right has noticeable circular distortions around the edge of the frame.
- I used the included demo application (depthai_demo.py) to verify this behavior.

### **My calibration board:**

- [Charuco calibration image](https://github.com/luxonis/depthai/blob/2402db26408da6a122d9ae9ae646b0d96ea7e1d9/charuco_11x8.pdf) printed on an A4 paper and glued perfectly flat to a rigid surface.

Erol444, 2022-12-23:
Hi @ZumWoelkchen , I think this is due to current camera model that we are using, which causes such undistortion artifacts. cc @saching13 to confirm.

saching13, 2022-12-24:
Yes. It is coming from RATIONLA POLYNOMIAL model. We will be changing it soon. Sorry for the issue.