luxonis/depthai #883: [BUG] Cannot use different OAK devices consecutively on the same USB port

brmarkus, 2022-12-22:
Currently setting-up a new Laptop with MS-Windows 11.

Installed DepthAI demo "v3.2.1" successfully via the Windows-Installer-Executable "DepthAI-setup-v3.2.1.exe" (in a CMD prompt with admin-rights, and setting network proxies first).

Now connecting OAK-D to a USB-3 port of the laptop, using the original USB3-USBC cable - demo is working fine!
Exiting the demo.
Then unplugging OAK-D and connecting OAK-1 on the same port, using the same cable, starting the demo - demo is working fine!
Exiting the demo.
Then unplugging OAK-1 and connecting OAK-D-Lite, using the same cable, connecting to the same USB3 port. But the demo fails with an exception:

```
---------------------------
An error occured
---------------------------
  File "C:\Users\me\AppData\Local\Programs\DepthAI\depthai\depthai_demo.py", line 655, in run
    self.instance.run_all(self.conf)
  File "C:\Users\me\AppData\Local\Programs\DepthAI\depthai\depthai_demo.py", line 131, in run_all
    self.setup(conf)
  File "C:\Users\me\AppData\Local\Programs\DepthAI\depthai\depthai_demo.py", line 208, in setup
    self._device = dai.Device(self._pm.pipeline.getOpenVINOVersion(), self._deviceInfo, usb2Mode=self._conf.args.usbSpeed == "usb2")
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
---------------------------
OK   
---------------------------
```

Getting these log messages in the CMD shell I started the demo in:

```
========= Starting: Launcher (2022-12-22 17:26:12.372146) =========
System information:
 - sys.prefix = C:\Users\me\AppData\Local\Programs\DepthAI\venv
 - sys.base_prefix = C:\Users\me\AppData\Local\Programs\DepthAI\WPy64-3950\python-3.9.5.amd64
 - Env 'PYTHONPATH': None
 - Executable: C:\Users\me\AppData\Local\Programs\DepthAI\venv\Scripts\python.exe

HEAD detached at v3.2.1
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        deleted:    launcher/windows/build.ps1
        deleted:    launcher/windows/download_dependencies.ps1
        deleted:    launcher/windows/inno_setup.ps1
        deleted:    launcher/windows/installer_win64.iss
        deleted:    launcher/windows/src/create_shortcut.ps1
        deleted:    launcher/windows/src/prerequisite.ps1
        deleted:    launcher/windows/version.txt

no changes added to commit (use "git add" and/or "git commit -a")
Current commit: 98bb628bb82da3f7396e086aa79647de6124ebab
Current tag: v3.2.1
Available DepthAI versions: ['v0.3.0.0', 'v0.4.0.0', 'v0.4.1.1', 'v1.0.0', 'v1.0.0.0', 'v2.0.0', 'v2.10.0.0', 'v2.7.2.0', 'v2.8.0.0', 'v2.9.0.0', 'v3.0.0', 'v3.0.1', 'v3.0.10', 'v3.0.11', 'v3.0.12', 'v3.0.12', 'v3.0.2', 'v3.0.3', 'v3.0.4', 'v3.0.5', 'v3.0.6', 'v3.0.7', 'v3.0.7', 'v3.0.8', 'v3.0.9', 'v3.1.0', 'v3.1.0', 'v3.1.1', 'v3.1.1', 'v3.1.2', 'v3.1.2', 'v3.1.3', 'v3.1.3', 'v3.2.0', 'v3.2.1']
Current tag: v3.2.1, ver: 3.2.1
Using depthai module from:  C:\Users\me\AppData\Local\Programs\DepthAI\venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.19.0.0
Setting up demo...
Available devices:
[0] 1944301071DBF01200 [X_LINK_UNBOOTED]
```

In "Device Manager" I do see the OAK-D-Lite device initially as "Movidius MyriadX" before starting the demo; then when the demo is initializing the USB-device disappears and doesn't appear again, as the assertion mentions.
After the assertion-message has been printed - the USB-device appears again as "Movidius MyriadX".
using OAK-D-Lite on another USB3, then the device appears as expected as "Luxonis Device".

Using OAK-D or OAK-1 with the same cable on the same port - and the demo is working again.

But now for OAK-D-Lite I need to use another USB3 port on my Laptop to get the demo to work.

Could that be MS-Win11 specific?
Could the previously detected USB-device information of OAK-D and OAK-1 be "reset" or "removed" ("forgotten") so that OAK-D-Lite will be detected and enumerated correctly on the same port again?

(I haven't tried such sequences of the OAK devices under MS-WIN10 on my previous Laptop, as far as I remember)

themarpe, 2022-12-24:
Thanks for the detailed breakdown @brmarkus 

Did you also test just plugging in OAK-D Lite to the same port but first? Does it work in that case? Is that the only port that doesn't work for Lite & if so, does it always not work?

> Could that be MS-Win11 specific?

Yeah, it might be but the bug itself could be in device handling, rather than Win11.

>  Could the previously detected USB-device information of OAK-D and OAK-1 be "reset" or "removed" ("forgotten") so that OAK-D-Lite will be detected and enumerated correctly on the same port again?

That depends on details from the first question - whether or not that port never works or only after OAK-1/D was used before it.

> (I haven't tried such sequences of the OAK devices under MS-WIN10 on my previous Laptop, as far as I remember)

It'd make a lot of sense trying this out on some other host as well. Let me know your findings

brmarkus, 2023-01-10:
My (new) laptop has very few USB-ports only; only one port is left, the others are in use (and I don't use a USB-hub).
That's why I connected the OAK devices one after the other to the same port.

I started testing OAK-D, followed by OAK-1, followed by OAK-D-Lite, in this random order.

However, in whatever order I connect the three devices to this one port only OAK-D and OAK-1 are detected and working with the demo; it is only the OAK-D-Lite which is not detected properly with this port.
Even after cold-boot it is the same behavior.
When I unplug the OAK-D-Lite and connect either OAK-D or OAK-1, then the two devices are working fine.

When I connect the OAK-D-Lite to the docking-station's USB-port, then OAK-D-Lite is detected and working properly.
When I unplug any other of my other USB-devices (like wireless headset Bluetooth dongle, mobile-phone) from the other Laptop's USB-ports and connect the OAK-D-Lite to that port, then the OAK-D-Lite is working.

It looks like USB-topology or USB-configuration information is stored on the Laptop persistently, and I was wondering if I can clear that info.
(I don't want to re-install MS-Win11 on that Laptop, and don't have another fresh Laptop available).

themarpe, 2023-01-10:
@brmarkus 

Do you mind testing the following:
1. Open Device Manager Window
2. Plug in OAK-D Lite into the "faulty" USB port
3. Check if Device Manager list refreshes.
4. If yes, check what new device popped up
5. if no, try rotating the USB-C port and plugging it again. Check step 4. again

Otherwise I don't have much more ideas on what could be the underlying issue here. 
Is the USB port on the laptop an A or Type-C port?

Is it Thunderbolt 3/4 capable port? Is that option in BIOS enabled/disabled?

brmarkus, 2023-01-12:
The port on the Laptop is an USB-Type-A port, USB3 superspeed capable.
Using Type-C to Type-A cable.
When connecting OAK-D Lite, then the DeviceManager shows "Movidius MyriadX".
When starting the demo, the device disappears - and appears again after ~10 seconds - instead of appearing as a "Luxonis Device" device.
Rotating the Type-C part of the cable at the OAK-D Lite side doesn't change the behavior.

Will just use another USB-Type-A connector from the laptop (re-arranging other cables, using ports on the docking-station) and all three devices (OAK-D, OAK-1, OAK-D Lite) are working using the same port.

closed 2023-01-12 by brmarkus