luxonis/depthai #886: [BUG] Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

DOKoenegras, 2022-12-28:
**Check if issue already exists**

There are multiple posts related but I couldn't resolve my issue. I've already spent hours on it but nothing worked so far...

https://github.com/luxonis/depthai/issues/753
https://github.com/luxonis/depthai-ros-examples/issues/57
https://github.com/luxonis/depthai/issues/518
https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html

**Describe the bug**
Failed to find device after booting.

**Hardware:**
OAK-D POE

**Host:** 
Raspberry 3b+

**Host OS (clean installation):**

PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

**Network interfaces:**

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.173.218  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::d92f:8d6f:bde:4e28  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:05:07:a9  txqueuelen 1000  (Ethernet)
        RX packets 80093  bytes 5343430 (5.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 160194  bytes 240715049 (229.5 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 573  bytes 19616 (19.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 573  bytes 19616 (19.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.129  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::e3:3e7:ea13:b443  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:50:52:fc  txqueuelen 1000  (Ethernet)
        RX packets 78294  bytes 106818769 (101.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 24604  bytes 1962901 (1.8 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**Steps to reproduce**

1. Installed the host OS 
2. Installed Depth AI via: sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash
3. Cloned depthai-python repo and installed requirements
5. Ran: python3 examples/ColorCamera/rgb_preview.py

**Output:**

das@raspberrypi:~/Luxonis/depthai-python $ DEPTHAI_LEVEL=debug python3 examples/ColorCamera/rgb_preview.py
[2022-12-28 10:53:12.897] [debug] Python bindings - version: 2.19.1.0 from  build: 2022-11-13 22:30:51 +0000
[2022-12-28 10:53:12.897] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-11-13 21:33:20 +0000
[2022-12-28 10:53:12.903] [debug] Initialize - finished
[2022-12-28 10:53:13.125] [debug] Device - OpenVINO version: 2022.1
[2022-12-28 10:53:13.125] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be be
[2022-12-28 10:53:13.208] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 6ms, archive read: 303ms
[2022-12-28 10:53:14.456] [debug] Resources - Archive 'depthai-device-fwp-adbcc016c8bd5a5580a26d8b6250f77160203666.tar.xz' open: 6ms, archive read: 1551ms
[2022-12-28 10:53:14.600] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=18443010F18EF80F00, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2022-12-28 10:53:14.826] [debug] Connected bootloader version 0.0.21
[2022-12-28 10:53:17.000] [debug] Booting FW with Bootloader. Version 0.0.21, Time taken: 2174ms
[2022-12-28 10:53:17.000] [debug] DeviceBootloader about to be closed...
[2022-12-28 10:53:17.001] [debug] XLinkResetRemote of linkId: (0)
[2022-12-28 10:53:17.827] [debug] DeviceBootloader closed, 826
[2022-12-28 10:53:17.856] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=18443010F18EF80F00, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
Traceback (most recent call last):
  File "/home/das/Luxonis/depthai-python/examples/ColorCamera/rgb_preview.py", line 24, in <module>
    with dai.Device(pipeline) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

**Additional notes**
- The camera has been factory reset via the device manager over USB.
- I can ping the camera on 169.254.1.222.
- The camera and host are directly connected via an ethernet cable and poe injector in between when the error occurs. When I connect them via my router everything works fine so it must be network related.
- I haven't made any network configuration changes.

Erol444, 2022-12-28:
Hi @DOKoenegras ,
I think the host computer (RPI) needs to be in the same lan eg. `169.254.1.X` so depthai can automatically find the OAK POE, or you need to [specify the IP](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/getting-started-with-poe.html#manually-specify-device-ip). Could you try that?
Thanks, Erik

DOKoenegras, 2022-12-28:
Hi @Erol444,

I've used the script to manually specify the ip. No luck so far. Any ideas?

The output:

das@raspberrypi:~ $ DEPTHAI_LEVEL=debug python3 test.py 
[2022-12-28 12:07:36.316] [debug] Python bindings - version: 2.19.1.0 from  build: 2022-11-13 22:30:51 +0000
[2022-12-28 12:07:36.316] [debug] Library information - version: 2.19.1, commit:  from , build: 2022-11-13 21:33:20 +0000
[2022-12-28 12:07:36.322] [debug] Initialize - finished
[2022-12-28 12:07:36.326] [debug] Found an actual device by given DeviceInfo: DeviceInfo(name=169.254.1.222, mxid=18443010F18EF80F00, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS)
[2022-12-28 12:07:36.326] [debug] Device - OpenVINO version: 2022.1
[2022-12-28 12:07:36.326] [debug] Device - BoardConfig: {"emmc":null,"gpio":[],"logDevicePrints":null,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 0e b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be be
[2022-12-28 12:07:36.620] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.21.tar.xz' open: 7ms, archive read: 296ms
[2022-12-28 12:07:37.854] [debug] Resources - Archive 'depthai-device-fwp-adbcc016c8bd5a5580a26d8b6250f77160203666.tar.xz' open: 6ms, archive read: 1530ms
[2022-12-28 12:07:37.991] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=18443010F18EF80F00, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2022-12-28 12:07:38.218] [debug] Connected bootloader version 0.0.21
[2022-12-28 12:07:40.393] [debug] Booting FW with Bootloader. Version 0.0.21, Time taken: 2174ms
[2022-12-28 12:07:40.393] [debug] DeviceBootloader about to be closed...
[2022-12-28 12:07:40.394] [debug] XLinkResetRemote of linkId: (0)
[2022-12-28 12:07:41.219] [debug] DeviceBootloader closed, 825
[2022-12-28 12:07:41.243] [debug] Searching for booted device: DeviceInfo(name=169.254.1.222, mxid=18443010F18EF80F00, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
Traceback (most recent call last):
  File "/home/das/test.py", line 16, in <module>
    with dai.Device(pipeline, device_info) as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

themarpe, 2022-12-28:
@DOKoenegras 
Interesting case - do you mind running list_devices.py in the background to see what happens after the PoE gets booted with FW.

If you use device manager to set a static IP, (eg something different than 169.254.1.222) IMO the problem shouldn't appear

themarpe, 2023-03-23:
@Ralbot 
Thanks for the update - do you mind letting me know what static IP you've assigned to the device?

jinkum, 2023-04-13:
I have same issue.

Device: Luxonis PRO POE 
Error message from "camera_preview.py" - "No OAK device found to connect to!"

I try to specify IP function from "device_manager.py" 
Error message: "failed to find device after booting. error message :X_LINK_DEVICE_NOT FOUNT"

OS: windows 11 pro

Note: last week I tested using same pc. But now not working.

Erol444, 2024-05-27:
Hi @Pannu22 ,
Could you try these:
- Run the script outside of the container (to determine this isn't the source of the issue)
- Update depthai to the latest version (2.26 at the moment)
- Update bootloader to the latest version as well (0.28)

Pannu22, 2024-05-29:
Hi @Erol444,

Thank you for getting back to me so quickly and for the suggested steps. While I appreciate your help, making the necessary changes to the client's setup is taking longer than expected because of their internal processes.

I'm so sorry for the delay and will follow up with you in a couple of days once I have access to their environment.

Thank you for your patience and understanding.

Pannu22, 2024-06-04:
Hi @Erol444 ,
Thank you for your solution. I tried it out, and it worked perfectly! I appreciate you taking the time to help resolve my issue.

closed 2024-06-04 by Erol444