luxonis/depthai #888: Encountering problems installing DepthAI...

coachroebuck, 2022-12-30:
I tried to run this command on multiple machines:
```
bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)"
```
I am running on a Macbook Pro, Ventura 13.0.1 (22A400)
```
Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 227 
```
Please advise as to any specifications and additional information you need from me in order to help me troubleshoot this problem. Thanks

themarpe, 2023-01-02:
Thanks for the report @coachroebuck 

@PetrNovota do you mind addressing?

PetrNovota, 2023-01-03:
@coachroebuck I am on it, I will get back to you very soon

PetrNovota, 2023-01-03:
@coachroebuck 
please send me last 10 lines of terminal output before you get the
`Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 227 `

coachroebuck, 2023-01-04:
Hi @PetrNovota ...

As crazy as this sounds, I'm not able to reproduce that error a second time, on either of these machines that I have. 

Since I'm unable to reproduce this issue, I will close this ticket.

Thanks!

closed 2023-01-04 by coachroebuck

PetrNovota, 2023-01-04:
@coachroebuck I reproduced same error you sent me. If "**brew upgrade**" does not work for any reason E.g. brew cannot find some formula because it was deleted manually etc.., then the demo installation fails with error you provided

coachroebuck, 2023-01-05:
Hi...

I see that same thing now. That makes sense.

themarpe, 2023-01-07:
@PetrNovota 

This one slipped my review

```
echo "Upgrading brew."
brew upgrade
```

According to docs this upgrades all packages. If that really is the case, and only brew update was intended, I'd rather modify this to do so. (otherwise system is modified too much IMO)

Seems like to upgrade brew itself, the following should be done:
```
brew update
```