luxonis/depthai #892: [BUG] Pipeline not working: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'

Thomas-C258, 2023-01-03:
**Check if issue already exists**

Similar to this: [#467 ]https://github.com/luxonis/depthai-core/issues/467
But I checked that my script is running on CSS, not MSS. And it already runs with a while loop.

**Describe the bug**
Two seconds in the execution of the pipeline, I get this error message: 
[1844301031690E1300] [3.8] [2.077] [system] [critical] Fatal error. Please report to developers. Log: 'Fatal error on MSS CPU: trap: 00, address: 00000000' '0'
No frames are received from the color camera through the preview xlinkout node. The immediate interruption makes it difficult to debug further on my own, especially when I have no idea what goes on on MSS.

**Minimal Reproducible Example**

[MRE.zip](https://github.com/luxonis/depthai/files/10335738/MRE.zip)

**Expected behavior**
My pipeline starts with a spatial ObjectTracker on RGB, and then for each new object tracked, tries to match it to stored references. Basically adding the person-reid model after the object tracker, and trying not to run the reid on every frame, but only on new objects or retry still unidentified objects after a while.

**Screenshots**
-

**Pipeline Graph**
[image]

**Attach system log**
 Log is in the MRE.zip. Working on windows 10, OAK-D-lite usb.

Thomas-C258, 2023-01-08:
@themarpe I saw that you labeled this issue as "oak d lite" specific, does that mean you managed to run it on another hardware? That'd be uplifting at least, knowing my code runs ! I can provide a pipeline that uses similar nodes, that is a tad more complex, and does the "initialisation" part of this one. It runs on my oak d lite, with sporadic CSS crashes though. You tell me if that can be useful.

themarpe, 2023-01-11:
@Thomas-C258 

I haven't been able to run it "successfully", but it didn't crash.

For the latter, double check you are on latest 2.19.1 version and supply a MRE please. Will help a lot

Thomas-C258, 2023-01-27:
[initialisation.zip](https://github.com/luxonis/depthai/files/10519535/initialisation.zip)
So I finally circled back to this bug. Here is the initialisation script, that does run and provides the expected json outputs. I do get the same MSS bug sometimes too, but it's sporadic and more rare (1 in 10?). It may be small sample, but it seems that sometimes unplugging the USB cable helps make the next run go smoothly.