luxonis/depthai #896: [BUG] {Title of the bug}

Shift-Ronan, 2023-01-09:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
I'm doing some development with the depthai-sdk package. When running the mcap-record.py script I get `NameError: name 'XoutFrames' is not defined` error from depthai_sdk/recorders/mcap_recorder.py. Before this, I was also missing the geometry_msgs package (and others) that I installed manually from https://rospypi.github.io/simple/.

**Minimal Reproducible Example**
Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)
I'm running depthai v2.19.1 on a Raspberry Pi 4 with Ubuntu 20.04 and followed these instructions for a Ubuntu installation:
https://docs.luxonis.com/projects/api/en/latest/install/
The script: https://github.com/luxonis/depthai/blob/main/depthai_sdk/src/depthai_sdk/recorders/mcap_recorder.py

**Expected behavior**
Depth and video data should be saved as MCAP file type.

**Screenshots**
> ~/Luxonis/depthai/depthai_sdk/examples/recording$ python mcap-record.py 
Recording...
Closing OAK camera
Traceback (most recent call last):
  File "mcap-record.py", line 12, in <module>
    recorder = oak.record([color.out.encoded, left, right, stereo.out.depth], './records/mcap', RecordType.MCAP)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/depthai_sdk/oak_camera.py", line 407, in record
    record = Record(Path(path).resolve(), record_type)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/depthai_sdk/record.py", line 78, in __init__
    from .recorders.mcap_recorder import McapRecorder
  File "/home/ubuntu/.local/lib/python3.8/site-packages/depthai_sdk/recorders/mcap_recorder.py", line 10, in <module>
    class McapRecorder(Recorder):
  File "/home/ubuntu/.local/lib/python3.8/site-packages/depthai_sdk/recorders/mcap_recorder.py", line 20, in McapRecorder
    def update(self, path: Path, device: dai.Device, xouts: List[XoutFrames]):
NameError: name 'XoutFrames' is not defined

**Pipeline Graph**
Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).
[image]
Had to use oak.show_graph() as the pipeline_graph script wasn't working. Commented out recording line so it didn't crash.
> Nodes (id):
===========
ColorCamera (0)
VideoEncoder (1)
MonoCamera (2)
MonoCamera (3)
StereoDepth (4)<br>
Connections:
============
0 MonoCamera (3): out -> StereoDepth (4): [8] right
1 MonoCamera (2): out -> StereoDepth (4): [8] left
2 ColorCamera (0): video -> VideoEncoder (1): [4] in

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
[log_system_information.txt](https://github.com/luxonis/depthai/files/10375619/log_system_information.txt)
Output as .txt because Markdown doesn't support uploading JSON.

**Additional context**
Unsure if it's related to this issue due to me having to manually install the ROS message definitions, but recording depth as rosbag files works but I'm unable to open them in the Intel Realsense Viewer due to a message error.

Erol444, 2023-03-01:
Has this been resolved @daniilpastukhov ?

daniilpastukhov, 2023-03-01:
Yes, it's fixed in `develop` branch already. The issue will be fixed in 1.9.5 that is coming soon