luxonis/depthai #897: Encounter issue during installation on Ubuntu 18.04

290811, 2023-01-10:
Hi, I encounter issue during installation on Ubuntu 18.04. When running the command " sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash" it fails with the following code:

     ERROR: Could not build wheels for sip, which is required to install pyproject.toml-based projects
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 222 

[image]

brmarkus, 2023-01-10:
Can you provide the full console output of the installation process, please?
Like around this snippet at least:
[image]
which seem to point to something.

PetrNovota, 2023-01-10:
@290811 please copy the entire terminal output and send it it us

RhythmDJay, 2023-01-27:
I am getting the same error trying to build on Jetpack 4.6.2 running Ubuntu 18.04. Here is the entire terminal output: ```sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash
[sudo] password for aaeon: 

Using default installation path: /home/aaeon/Luxonis
Directory: /home/aaeon/Luxonis is OK

Python version: Python 3.6.9 found.
This python version is not supported by depthai. Enter path to python binary version et least 3.8, or leave empty and python 3.10 will be installed automatically.
_____________________________
Calling linux_installer.sh
_____________________________
Updating sudo-apt.
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [83.3 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 Packages [1,731 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main Translation-en [527 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [291 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 48x48 Icons [83.0 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main DEP-11 64x64 Icons [154 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted arm64 Packages [5,360 B]
Get:11 http://ports.ubuntu.com/ubuntu-ports bionic-updates/restricted Translation-en [155 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 Packages [1,673 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe Translation-en [405 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [297 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 48x48 Icons [232 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe DEP-11 64x64 Icons [516 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse arm64 Packages [4,944 B]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-updates/multiverse Translation-en [6,088 B]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main arm64 Packages [53.2 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main Translation-en [14.6 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main arm64 DEP-11 Metadata [8,116 B]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main DEP-11 48x48 Icons [6,984 B]
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-backports/main DEP-11 64x64 Icons [10.5 kB]
Get:24 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 Packages [17.8 kB]
Get:25 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe Translation-en [8,668 B]
Get:26 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [10.0 kB]
Get:27 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe DEP-11 48x48 Icons [4,959 B]
Get:28 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe DEP-11 64x64 Icons [6,750 B]
Get:29 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 Packages [1,441 kB]
Get:30 http://ports.ubuntu.com/ubuntu-ports bionic-security/main Translation-en [440 kB]
Get:31 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [70.8 kB]
Get:32 http://ports.ubuntu.com/ubuntu-ports bionic-security/main DEP-11 48x48 Icons [31.1 kB]
Get:33 http://ports.ubuntu.com/ubuntu-ports bionic-security/main DEP-11 64x64 Icons [67.9 kB]
Get:34 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted arm64 Packages [4,808 B]
Get:35 http://ports.ubuntu.com/ubuntu-ports bionic-security/restricted Translation-en [150 kB]
Get:36 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 Packages [1,125 kB]
Get:37 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe Translation-en [291 kB]
Get:38 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [56.1 kB]
Get:39 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 48x48 Icons [31.3 kB]
Get:40 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe DEP-11 64x64 Icons [147 kB]
Get:41 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse arm64 Packages [2,668 B]
Get:42 http://ports.ubuntu.com/ubuntu-ports bionic-security/multiverse Translation-en [3,908 B]
Fetched 10.3 MB in 8s (1,255 kB/s)                                             
Reading package lists... Done
Installing global dependencies.
Hit:1 http://ports.ubuntu.com/ubuntu-ports bionic InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease
Reading package lists... Done                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
build-essential set to manually installed.
libsm6 is already the newest version (2:1.2.2-1).
libsm6 set to manually installed.
libxext6 is already the newest version (2:1.3.3-1).
[... 837 lines cut ...]
[... cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 222```

RhythmDJay, 2023-01-30:
I ended up with a low storage message.
Collecting pyqt5
  Using cached PyQt5-5.15.8.tar.gz (3.2 MB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [4 lines of output]
      Collecting sip<7,>=6.6.2
        Using cached sip-6.7.6.tar.gz (1.2 MB)
      ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device

PetrNovota, 2023-01-30:
@RhythmDJay thank you. This looks like a good progress. Could you please delete some data on your hard drive to free up space? I would say that clearing approx. 200MB of data should be enough. Then do the above mentioned steps again. If you obtain the `ERROR: Could not install packages due to an OSError: [Errno 28] No space left on device` error, clear more space.
If the installation completes, try the depthai installation again, i.e. `sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash`
Please let me know how that goes no matter the result

PetrNovota, 2023-01-30:
@RhythmDJay I think I have found the problem, the script used `pip install --upgrade pip` instead of `python -m pip install --upgrade pip`. Got fixed in [PR](https://github.com/luxonis/depthai-python/pull/755).
If you encounter any problems, let us know. Please create new issue for that as I consider this one fixed

RhythmDJay, 2023-02-02:
Hi @PetrNovota it looks like after clearing space I'm getting the original issue again. 

I get: 
`ERROR: Failed building wheel for sip`

Anything else you can think of?

PetrNovota, 2023-02-02:
@RhythmDJay thank you for the message.
Just to clarify. You are using a Jetson platform which is using ARM CPU?

The error comes from installing PyQt5 which is a package for developing front end. It is widely used. The sip package that you cannot install in thy PyQt5 dependency. 

I dont have the Jetson platform available so I cannot recreate the problem you are having.

We just have to figure out how to build the sip package from source. Please try the following:

1. open new terminal window
2. run: `sudo apt-get install python3.10-dev`
3. activate virtual environment with `source /home/aaeon/Luxonis/venv/bin/activate`
4. run: `python -m pip install --upgrade pip setuptools wheel`
5. run: `pip install sip`

Please let me know how it goes. this will either result in the same error as before or sip will get installed, if it gets installed you may try the depthai installation once more.

themarpe, 2023-02-02:
@PetrNovota PyQt5 doens't provide wheels for AArch64 nor armv7l. Its common that those are used "from the system", by eg `sudo apt install python3-pyqt5`. I think this would be best path for that platform. Note, in that case the installation of custom python (3.10) should be skipped

PetrNovota, 2023-02-03:
@themarpe his global python interpreter is v. 3.6.9. Is it fine for depthai or should we update?

themarpe, 2023-02-04:
From core's perspective, yes

PetrNovota, 2023-02-06:
@RhythmDJay I think we have a good plan how to make it work for you. If you dont mind I would start by installing the Pyqt5 as my college suggested. Please open a terminal and type `sudo apt install python3-pyqt5`. 
Then continue with the following:

1. delete `/home/aaeon/Luxonis/venv` directory  # we will install new one
2. cd /home/aaeon/Luxonis
3. sudo pip3 install virtualenv  # making sure virtualenv is installed. might be already installed
4. virtualenv --system-site-packages venv  # pyqt5 is installed as global package. This ensures your venv inherits those as well
5. source venv/bin/activate  # activate virtual environment
6. pip install -I -r depthai/install_requrements.txt  # install depthai dependencies
7. python depthai/depthai_demo.py  # run demo. Make sure you have a OAK device connected at this point

Let me know how it goes.