luxonis/depthai #904: [BUG] OV9782 Variants of POE devices Issue with tracking node

justin-larking-pk, 2023-01-15:
**Describe the bug**
OV9782 variant of the OAK-D-POE-PRO-W does not run pipelines contain the object tracker node. Meanwhile this exact same script and pipeline configuration runs without issue on the OAK-D-POR-POE-W with the imx378 sensor. I have tested this pipeline on four different depthai devices.

As above the OAK-D-PRO-W-POE (OV9782): Does not run halts on the tracker output "q_tracks.get()"
OAK-D-PRO-W-DEV: Does not run halts on the object tracker output
OAK-D(original variant not s2): Runs without issue
OAK-D-PRO-W-POE (imx378): Runs without issue

The only common denominator here seems to be the presence of the ov9782 sensor. In addition I believe its an issue with the tracking node as the script consistently halts on the output of the tracklet queue and other scripts without object tracking run fine on the OAK-D-PRO-W-POE (OV9782).

THere is also a MRE in c++ that can be provided. This was originally tested with depthai-core 2.19. 

**Minimal Reproducible Example**
Python MRE:
`
import depthai as dai
import cv2
import numpy as np

rgb_res = "720p"

def main():

    pipeline = dai.Pipeline()

    spatialdetectionnetwork = pipeline.create(dai.node.MobileNetSpatialDetectionNetwork)
    camrgb = pipeline.create(dai.node.ColorCamera)
    monoleft = pipeline.create(dai.node.MonoCamera)
    monoright = pipeline.create(dai.node.MonoCamera)
    depth = pipeline.create(dai.node.StereoDepth)
    imu = pipeline.create(dai.node.IMU)
    objectTracker = pipeline.create(dai.node.ObjectTracker)
    videoenc = pipeline.create(dai.node.VideoEncoder)

    xoutdepth = pipeline.create(dai.node.XLinkOut)
    xoutvideo = pipeline.create(dai.node.XLinkOut)
    xouttracks = pipeline.create(dai.node.XLinkOut)
    xoutimu = pipeline.create(dai.node.XLinkOut)
    xoutenc = pipeline.create(dai.node.XLinkOut)

    xouttracks.setStreamName("tracklets")
    xoutdepth.setStreamName("depth")
    xoutvideo.setStreamName("video")
    xoutimu.setStreamName("imu")
    xoutenc.setStreamName("enc")

    camrgb.setPreviewSize(300, 300)
    camrgb.setInterleaved(False)
    camrgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.BGR)
    if(rgb_res == "1080p"):
        camrgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
        camrgb.setIspScale(2, 3)
    else:
        camrgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_720_P)

    camrgb.setPreviewKeepAspectRatio(False)
    monoleft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    monoleft.setBoardSocket(dai.CameraBoardSocket.LEFT)
    monoright.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    monoright.setBoardSocket(dai.CameraBoardSocket.RIGHT)
    depth.setDefaultProfilePreset(dai.node.StereoDepth.PresetMode.HIGH_DENSITY)
    depth.initialConfig.setMedianFilter(dai.MedianFilter.KERNEL_7x7)
    depth.setLeftRightCheck(True)
    depth.setExtendedDisparity(True)
    depth.left.setBlocking(False)
    depth.right.setBlocking(False)
    depth.setDepthAlign(dai.CameraBoardSocket.RGB)
    depth.setOutputSize(monoleft.getResolutionWidth(), monoleft.getResolutionHeight())
    depth.useHomographyRectification(False)
    max_disp = depth.initialConfig.getMaxDisparity()
[... 589 lines cut ...]
            "speed": "Full"
        },
        {
            "port": 13,
            "vendor_id": "0x17ef",
            "product_id": "0x60a9",
            "speed": "Full"
        },
        {
            "port": 12,
            "vendor_id": "0x03e7",
            "product_id": "0x2485",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ],
    "uname": [
        "Linux justin-ThinkStation-P720 5.14.0-1055-oem #62-Ubuntu SMP Wed Nov 30 04:54:03 UTC 2022 x86_64 x86_64"
    ]
}

szabi-luxonis, 2023-01-16:
Hello @justin-larking-pk 
To run object tracker example checkout latest `main`, run `examples/install_requirements.py` and run one of the examples:
[spatial_object_tracker.py](https://github.com/luxonis/depthai-python/blob/main/examples/ObjectTracker/spatial_object_tracker.py)
[object_tracker.py](https://github.com/luxonis/depthai-python/blob/main/examples/ObjectTracker/object_tracker.py)

They run as it is.
If you want to get rid of the error message modify the following line:
```
camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_720_P)
```

justin-larking-pk, 2023-01-16:
The spatial_object_tracker.py does indeed run on my OAK-D-W-PRO-POE w/ OV9782 also. I guess this eliminates the object tracker node as being the issue.

However the issue still exists as some OAK-D cameras can run that script I posted above and some cannot.

justin-larking-pk, 2023-01-24:
@szabi-luxonis Have you attempted to run my posted example? It may not be that tracking as I thought was breaking it. However this pipeline still only runs on certain depthai cameras, and thus is still an issue.

szabi-luxonis, 2023-01-24:
@justin-larking-pk can you try updating the depthai lib to the latest? We had some bug fixes recently.

justin-larking-pk, 2023-01-24:
Updated from depthai on pip 2.19.1 to 2.20.1, no difference still halts on oakd-poe-pro-w (ov9782)

justin-larking-pk, 2023-01-25:
Is there no one at Luxonis who can simply run that example that is causing this discrepancy on another ov9782 device and another imx378 one, as a sanity check? I have tried every device I have, the behaviour is exactly the same. The script runs fine on cameras with imx378 sensor, and hatls on ones with the ov9782, this is the only cmmon difference between the cameras I have tested.

szabi-luxonis, 2023-01-25:
The above pipeline works for me @justin-larking-pk :

[image]

szabi-luxonis, 2023-01-25:
@justin-larking-pk can you set: `export DEPTHAI_DEBUG=1` environment variable and post the logs?

justin-larking-pk, 2023-01-25:
[2023-01-26 10:46:48.185] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-01-26 10:46:48.185] [debug] Python bindings - version: 2.20.1.0 from  build: 2023-01-24 16:15:18 +0000
[2023-01-26 10:46:48.185] [debug] Library information - version: 2.20.1, commit: fb873489dba2185f74b4bb95e0ae1b163a386720 from 2023-01-24 16:56:47 +0200, build: 2023-01-24 16:03:12 +0000
[2023-01-26 10:46:48.186] [debug] Initialize - finished
[2023-01-26 10:46:48.387] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.24.tar.xz' open: 9ms, archive read: 192ms
[2023-01-26 10:46:48.741] [debug] Device - OpenVINO version: 2022.1
[2023-01-26 10:46:48.742] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
libnop:
0000: b9 10 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
0020: be be be 01 00 bb 00
[2023-01-26 10:46:48.856] [debug] Resources - Archive 'depthai-device-fwp-f888710ca677ca662a4a83103e57c5a1ae2a5c7f.tar.xz' open: 10ms, archive read: 660ms
[2023-01-26 10:46:48.902] [debug] Searching for booted device: DeviceInfo(name=192.168.1.129, mxid=194430106195741300, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-01-26 10:46:48.972] [debug] Connected bootloader version 0.0.19
[2023-01-26 10:46:48.972] [info] New bootloader version available. Device has: 0.0.19, available: 0.0.24
[2023-01-26 10:46:49.828] [debug] Booting FW with Bootloader. Version 0.0.19, Time taken: 856ms
[2023-01-26 10:46:49.828] [debug] DeviceBootloader about to be closed...
[2023-01-26 10:46:49.829] [debug] XLinkResetRemote of linkId: (0)
[2023-01-26 10:46:50.472] [debug] DeviceBootloader closed, 643
[2023-01-26 10:46:50.477] [debug] Searching for booted device: DeviceInfo(name=192.168.1.129, mxid=194430106195741300, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[194430106195741300] [192.168.1.129] [4.898] [system] [warning] PRINT:LeonCss: GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
====ENABLE WATCHDOG====1
initial keepalive, countdown: 7
PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
Board init ret 3
eeprom configuration version: 55AA0007
Reading VERSION 7 --- ->
eeprom configuration load from user area, status: 0
Reading VERSION 7 --- ->
Found a device/board entry matching the eeprom data - Board: OAK-D S2 PoE (generic) Device: 
Board options: 00000007
--> brdInit ...
brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]

spi_N25Q_init: Flash JEDEC ID: c2 25 3a
SR CR: 40 07
QUAD mode already enabled. Dummy cfg: 0x00
Opening bus for IR driver: 1
Failed to probe IR driver LM3644
Opening bus for IR driver: 2
LM3644 detected, ID = 0x02
===== IR write bus 2: 0x07 = 0x89
===== IR write bus 2: 0x02 = 0x01
===== IR write bus 2: 0x03 = 0x00
===== IR write bus 2: 0x04 = 0x00
===== IR write bus 2: 0x05 = 0x00
===== IR write bus 2: 0x06 = 0x00
===== IR write bus 2: 0x07 = 0x09
===== IR write bus 2: 0x08 = 0x1a
===== IR write bus 2: 0x09 = 0x08
===== IR write bus 2: 0x01 = 0x24
Opening bus for IR driver: 3
Failed to probe IR driver LM3644
Closing EEPROm!
MyriaX board configuration
pll0 frequency: 700000, ref0 frequency: 24000
Setup network with bootloader config
OsDrvCprSysDeviceAction() for PCIe failed: 13
nexus0: <RTEMS Nexus device>
pcib0: <MyriadX PCIe RC> on nexus0
PCIE_REGS_PCIE_CFG_ADR = 0x60100
PCIE_REGS_PCIE_PLL_CNTRL_ADR = 0x1
pcib0: Link training complete, PCIe downstream link is UP
pcib0: myriad_pcie_err_interrupt: SMLH LINK UP
pcib0: myriad_pcie_err_interrupt: SMLH LINK DOWN
pcib0: myriad_pcie_err_interrupt: RDLH LINK UP
pcib0: myriad_pcie_err_interrupt: RDLH LINK DOWN
pcib0: myriad_pcie_err_interrupt: CORE RST
pcib0: myriad_pcie_err_interrupt: CORE RSTN
pcib0: myriad_pcie_err_interrupt: LINK REQ RST
[... 251 lines cut ...]
[194430106195741300] [192.168.1.129] [8.766] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [8.832] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [8.898] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [8.920] [system] [info] Memory Usage - DDR: 78.87 / 340.42 MiB, CMX: 2.40 / 2.50 MiB, LeonOS Heap: 66.22 / 77.33 MiB, LeonRT Heap: 6.51 / 41.23 MiB
[194430106195741300] [192.168.1.129] [8.920] [system] [info] Temperatures - Average: 41.84 °C, CSS: 42.35 °C, MSS 41.89 °C, UPA: 42.12 °C, DSS: 40.97 °C
[194430106195741300] [192.168.1.129] [8.920] [system] [info] Cpu Usage - LeonOS 43.61%, LeonRT: 1.72%
[194430106195741300] [192.168.1.129] [8.964] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [9.030] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [9.096] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[194430106195741300] [192.168.1.129] [9.162] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!
[... cut ...]
[194430106195741300] [192.168.1.129] [9.922] [system] [info] Memory Usage - DDR: 78.87 / 340.42 MiB, CMX: 2.40 / 2.50 MiB, LeonOS Heap: 66.22 / 77.33 MiB, LeonRT Heap: 6.51 / 41.23 MiB
[194430106195741300] [192.168.1.129] [9.922] [system] [info] Temperatures - Average: 42.12 °C, CSS: 42.81 °C, MSS 41.20 °C, UPA: 42.58 °C, DSS: 41.89 °C
[194430106195741300] [192.168.1.129] [9.922] [system] [info] Cpu Usage - LeonOS 43.60%, LeonRT: 1.76%
[194430106195741300] [192.168.1.129] [9.965] [system] [warning] PRINT:LeonMss: stereo sync - missing frames!

justin-larking-pk, 2023-01-25:
I have removed the schema dump that shows the pipeline. Instead i'll give you the pipeline_graph output.
[image]

justin-larking-pk, 2023-01-25:
As the full dump is horrible to look at here is a screenshot of log output which contains errors.
[image]

szabi-luxonis, 2023-01-25:
@justin-larking-pk looks like there's a sync issue between stereo cameras, when 9782 is present. 
I will be able to look into it later this week.

szabi-luxonis, 2023-01-27:
We found the issue @justin-larking-pk.
WIll share the updated FW with the fix soon.

alex-luxonis, 2023-01-27:
Fix pushed to the branch `release_2.20.2` (https://github.com/luxonis/depthai-core/pull/694) and `develop`.

justin-larking-pk, 2023-01-29:
Awesome, tested the branch and its now working for me, thanks for the help.

closed 2023-01-29 by justin-larking-pk