luxonis/depthai #908: [BUG] OpenVino 2022.1 input shape problem API 2.0

dexter-8, 2023-01-19:
This came from https://discuss.luxonis.com/d/1073-blobconverter-issue

About the issue with API 2.0 in OpenVino 2022.1.

Files attached in the zip.

1) Get the blobs using the python files
2) Run the C++ code switching the code to use the blob from 2021.4 or 2022.1.

When running 2021.4 the log looks like this (as expected):

--- TEST ---
- depthai version 2.19.1
[194430100156E21200] [2.14] [0.958] [NeuralNetwork(2)] [warning] Number of inference threads assigned for network is 1, assigning 2 will likely yield in better performance
- mxid 194430100156E21200
- usb speed SUPER
- camera count 3
--------------

When running 2022.1 the log looks like this (not expected, should be like above):

--- TEST ---
- depthai version 2.19.1
[194430100156E21200] [2.14] [0.927] [NeuralNetwork(2)] [warning] Number of inference threads assigned for network is 1, assigning 2 will likely yield in better performance
- mxid 194430100156E21200
- usb speed SUPER
- camera count 3
[194430100156E21200] [2.14] [2.634] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [3.518] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [4.403] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [5.281] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [6.162] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [7.040] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [7.919] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [8.802] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [9.679] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)
-[194430100156E21200] [2.14] [10.562] [NeuralNetwork(2)] [warning] Input image (1280x768) does not match NN (3x1280)

When adding "-ov_api_1_0" to the advanced parameters for 2022.1 the NN runs without the warning (and behaves ok).

[luxonis-test.zip](https://github.com/luxonis/depthai/files/10455419/luxonis-test.zip)

Erol444, 2023-01-19:
Hi @dexter-8 ,
Perhaps you would need to specify the model input size/shape during the openvino model optimizer process? Also cc @tersekmatija as I think I have seen similar 2021.4 vs 2022.1 issue before.
Thanks, Erik

dexter-8, 2023-01-19:
I don't know if this is what you mean, but I tried options from [here](https://docs.openvino.ai/latest/openvino_inference_engine_tools_compile_tool_README.html) to change input and model parameters and no combination improved things, just different warnings.
There appears to be a similar issue with text-recognition-0012, mentioned [here](https://discuss.luxonis.com/d/929-text-recognition-0012-in-c).

tersekmatija, 2023-01-21:
Hey, it does seem like the input shape in OpenVINO model zoo was changed. [2021.4 version here](https://storage.openvinotoolkit.org/repositories/open_model_zoo/2021.4/models_bin/1/text-detection-0004/FP32/text-detection-0004.xml) expects [Batch,Channels,Height,Width] input, while [2022.1 version](https://storage.openvinotoolkit.org/repositories/open_model_zoo/2022.1/models_bin/1/text-detection-0004/FP32/text-detection-0004.xml) expects [B,H,W,C] input. When using the model from 2022.1 model zoo, DepthAI then causes the error you are seeing, which tells you that the input shape from device [1, C, H, W] does not match the input shape of the model. Using `-ov_api_1_0` actually forces OpenVINO under the hood to use 2021.4, which is the reason why this model works.

Since OpenVINO model zoo provides already exported models (xml and bin), you cannot provide additional shape flags and force it to use [B,C,H,W], so the only way is to use the model from 2021.4 ZOO. Another option we could expose, but would likely be inefficient and cause further confusion, is allowing images to be fed in [B,H,W,C] format on device (CC @themarpe). If you would be feeding data from the host, this should be relatively easy to do by sending a flattened [H,W,C] image with NNData.

dexter-8, 2023-01-22:
Thanks for investigating!
Since the frame rate is low I can try to reshape it on the host if I can fit it in without using more memory (already low on device memory).
I thought the connection between the camera and the network input actually checked the shape and adapted to it, but that doesn't seem to be the case. Only networks with BCHW can be directly connected to the camera then?
The [openvino page](https://docs.openvino.ai/latest/omz_models_model_text_detection_0004.html) states BHWC as well.

tersekmatija, 2023-01-23:
> I thought the connection between the camera and the network input actually checked the shape and adapted to it, but that doesn't seem to be the case.
Yes, my understanding is that we flatten the array accordingly on the DepthAI side which means that for non-image inputs shape data is lost and is being reshaped into the input shape. For image data we use ImgFrame object, where width and height are known, but we don't do dimension re-ordering. CC @themarpe if that's correct.

dexter-8, 2023-01-23:
I'm trying to send data to the network from the host in order to reshape the input. I can't figure out what I'm doing wrong though. I only get one output and then it stops (I keep sending input to the non blocking network queue). I didn't get sequenceNum in the network output so I added passthrough. The passthrough gives me the sequenceNum. The main change was to not route the camera preview to the network, but to send it through a queue instead.

tersekmatija, 2023-01-25:
Yeah, I assume the tensor is reshaped to the input shape. So if you have the same input shape on host, flatten, and send, it will shape it correctly (CC @themarpe ).

For blocking issues we'd need to see the pipeline. And then @Erol444 would be probably most experienced to advise why blocking is happening.

dexter-8, 2023-01-25:
I created a separate ticket [(917)](https://github.com/luxonis/depthai/issues/917) for the blocking issue with just one network, similar to the code here.

dexter-8, 2023-01-25:
It would be nice to have an option to set the output shape of the camera preview to make it more compatible with different networks.