luxonis/depthai #918: [BUG] bash: line 231: cho: command not found

Bengt, 2023-01-25:


Erol444, 2023-01-25:
Hi @Bengt  Could you perhaps provide some additional description that would help us debug this issue?
Thanks, Erik

Bengt, 2023-01-25:
Hi, yes, sorry. I was still debugging this. The error occurred when running the installer script:

    sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash

It worked anyway and I could manually run the application under `~/Luxonis`

Bengt, 2023-01-25:
Some googling points to possibly this meaning `echo`, but the `e` was somehow lost in translation somewhere. - Just an idea.

Erol444, 2023-01-25:
Hi @Bengt ,
Could you provide the full log you get when executing that command? What OS are you using?

[image]

Looking at https://docs.luxonis.com/install_depthai.sh it looks correct to me, not sure how it could come to parsing issues..
Thanks, Erik

Bengt, 2023-01-26:
This was under Ubuntu 22.04.

themarpe, 2023-01-26:
Thanks for the report @Bengt  its acutally inline with one other issue we had.

@PetrNovota do you mind searching through all use cases of the `$'` and replace with more appropriate equivalents?

PetrNovota, 2023-01-26:
@Bengt thank you for reporting this error. 
@themarpe I have created a PR.
 I have noticed another unwanted behavior. The script does not wait for user input. Lets see if that is also solved by this. 
Strange to me is that if I run the script locally it works as expected, but running it via `sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash` produces different and unwanted behavior

PetrNovota, 2023-01-27:
@themarpe the PR is done. read command did not work properly. Had to specify ` < /dev/tty`

closed 2023-01-30 by themarpe