luxonis/depthai #923: [BUG] Preview size error even if not using preview

domef, 2023-02-01:
**Describe the bug**
When streaming for example with `video` or `isp` outputs, an error on preview size is thrown, even if I am not using the `preview` output:

```
Traceback (most recent call last):
  File "program.py", line 18, in <module>
    device = dai.Device(pipeline, usb2Mode=False)
RuntimeError: ColorCamera(0) - 'preview' width or height (300, 300) bigger than sensor resolution (480, 270)
```

**Minimal Reproducible Example**
This is my code:

```python
import cv2
import depthai as dai

pipeline = dai.Pipeline()
camera = pipeline.create(dai.node.ColorCamera)
camera.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
camera.setIspScale(1, 4)
link = pipeline.createXLinkOut()
link.setStreamName("rgb")
link.input.setBlocking(False)
camera.video.link(link.input)
device = dai.Device(pipeline, usb2Mode=False)
queue = device.getOutputQueue(name="rgb", maxSize=10, blocking=False)
while True:
    image_raw = queue.get()
    image = image_raw.getCvFrame()
    cv2.namedWindow("image", cv2.WINDOW_NORMAL)
    cv2.imshow("image", image)
    c = cv2.waitKey(1)
    if c == ord("q"):
        break
```

The resolution 300x300 is the default preview size. To fix I have to add this line:

```
camera.setPreviewSize(2, 2)
```

**Expected behavior**
There shouldn't be a check on preview size if I'm not using `preview` output.

**Attach system log**
```json
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.0-58-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default Nov 14 2022 12:59:47",
    "python_compiler": "GCC 9.4.0",
    "python_implementation": "CPython",
    "python_version": "3.8.10",
    "release": "5.15.0-58-generic",
    "system": "Linux",
    "version": "#64~20.04.1-Ubuntu SMP Fri Jan 6 16:42:31 UTC 2023",
    "win32_ver": "",
    "packages": [
        "depthai==2.20.2.0",
        "numpy==1.24.1",
        "opencv-python==4.7.0.68",
        "pip==23.0",
        "pkg_resources==0.0.0",
        "setuptools==67.0.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux federico-eyecan 5.15.0-58-generic #64~20.04.1-Ubuntu SMP Fri Jan 6 16:42:31 UTC 2023 x86_64 x86_64"
    ]
}
```

Erol444, 2023-02-01:
Hi @domef ,
You are down-scaling the ISP output;
```
camera.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
camera.setIspScale(1, 4)
```
Which means it will go from 1920x1080 to 480x270. And since video/preview are derived from isp output, you can't have higher preview/video size than isp (in your case 480x270). So you shouldn't downscale the isp that much if you want to have 300x300 preview. Thoughts?
Thanks, Erik

domef, 2023-02-01:
The point is that I'm not using the preview, so why it should throw an error?

domef, 2023-02-01:
Moreover I just fount out that my "hack" can cause some problems (that doesn't appear without `camera.setPreviewSize(2, 2)`), in particular with this code:

```python
import cv2
import depthai as dai

pipeline = dai.Pipeline()
camera = pipeline.create(dai.node.ColorCamera)
camera.setPreviewSize(2, 2)
camera.setResolution(dai.ColorCameraProperties.SensorResolution.THE_13_MP)
camera.setIspScale(1, 4)
link = pipeline.createXLinkOut()
link.setStreamName("rgb")
link.input.setBlocking(False)
camera.video.link(link.input)
device = dai.Device(pipeline, usb2Mode=False)
queue = device.getOutputQueue(name="rgb", maxSize=10, blocking=False)
while True:
    image_raw = queue.get()
    image = image_raw.getCvFrame()
    print(image.shape)
    cv2.namedWindow("image", cv2.WINDOW_NORMAL)
    cv2.imshow("image", image)
    c = cv2.waitKey(1)
    if c == ord("q"):
        break
```
an error is continuosly thrown:

```
[18443010C1818F0E00] [1.1] [1.139] [ColorCamera(0)] [error] RGB postprocessing error, rescheduling!
[18443010C1818F0E00] [1.1] [1.245] [ColorCamera(0)] [error] RGB postprocessing timeout!
[18443010C1818F0E00] [1.1] [1.245] [ColorCamera(0)] [error] RGB postprocessing error, rescheduling!
[18443010C1818F0E00] [1.1] [1.311] [ColorCamera(0)] [error] RGB postprocessing timeout!
```