luxonis/depthai #928: [BUG] {ImportError: cannot import name 'FPSHandler' from 'depthai_sdk'}

AndrewBarfield, 2023-02-03:
On Ubuntu 22.04

Error launching demo:

`Traceback (most recent call last):
  File "/workdrive/Repositories/depthai/depthai_demo.py", line 54, in <module>
    from depthai_sdk import FPSHandler, loadModule, getDeviceInfo, downloadYTVideo, Previews, createBlankFrame
ImportError: cannot import name 'FPSHandler' from 'depthai_sdk' (/home/drew/.local/lib/python3.10/site-packages/depthai_sdk/__init__.py)
`

I've uninstalled and reinstalled everything to no avail.

compscirunner, 2023-02-04:
Having same issue, 
Traceback (most recent call last):
  File "/home/david/Luxonis/depthai/depthai_demo.py", line 54, in <module>
    from depthai_sdk import FPSHandler, loadModule, getDeviceInfo, downloadYTVideo, Previews, createBlankFrame
ImportError: cannot import name 'FPSHandler' from 'depthai_sdk' (/home/david/Luxonis/venv/lib/python3.10/site-packages/depthai_sdk/__init__.py)
^C
david@pop-os:~/davelab/depthai/depthai$ lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

daniilpastukhov, 2023-02-04:
Hi,
thanks you for reporting the problem. Addressed this issue in #929.
Daniil

closed 2023-02-04 by daniilpastukhov