luxonis/depthai #932: AttributeError: 'Demo' object has no attribute '_pm'

sivaparvathimannem, 2023-02-06:
I am trying to run the depthai_demo.py by following the steps from here https://docs.luxonis.com/projects/api/en/latest/install/
when i try to run the demo i am getting the following errors.  can someone help me.
Depthai version installed:  2.20.2.0
Traceback (most recent call last):
  File "depthai_demo.py", line 1051, in <module>
    runQt()
  File "depthai_demo.py", line 584, in runQt
    from gui.main import DemoQtGui
  File "/home/mannem/depthai/gui/main.py", line 16, in <module>
    from depthai_sdk import Previews, resizeLetterbox, createBlankFrame
ImportError: cannot import name 'Previews' from 'depthai_sdk' (/home/mannem/.virtualenvs/depthAI/lib/python3.8/site-packages/depthai_sdk/__init__.py)
Error while running demo script... Command '"/home/mannem/.virtualenvs/depthAI/bin/python" depthai_demo.py --noSupervisor --guiType qt' returned non-zero exit status 1.
Waiting 5s for the device to be discoverable again...
Setting up demo...
Traceback (most recent call last):
  File "depthai_demo.py", line 1054, in <module>
    runOpenCv()
  File "depthai_demo.py", line 1044, in runOpenCv
    demo.run_all(confManager)
  File "depthai_demo.py", line 133, in run_all
    self.setup(conf)
  File "depthai_demo.py", line 181, in setup
    self._deviceInfo = getDeviceInfo(self._conf.args.deviceId, args.debug)
  File "/home/mannem/.virtualenvs/depthAI/lib/python3.8/site-packages/depthai_sdk/utils.py", line 226, in getDeviceInfo
    raise RuntimeError("No DepthAI device found!")
RuntimeError: No DepthAI device found!
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "depthai_demo.py", line 335, in stop
    self._pm.closeDefaultQueues()
AttributeError: 'Demo' object has no attribute '_pm'

hlord2000, 2023-02-07:
Similar issue on Mint 21.1 
Kernel 5.15.5.0-58-generic
AMD Ryzen 7 5800X

> python3 depthai_demo.py 
> Using depthai module from:  /home/h/.local/lib/python3.10/site-packages/depthai.cpython-310-x86_64-linux-gnu.so
> Depthai version installed:  2.20.2.0
> Traceback (most recent call last):
>   File "/home/h/depthai/depthai_demo.py", line 1051, in <module>
>     runQt()
>   File "/home/h/depthai/depthai_demo.py", line 584, in runQt
>     from gui.main import DemoQtGui
>   File "/home/h/depthai/gui/main.py", line 16, in <module>
>     from depthai_sdk import Previews, resizeLetterbox, createBlankFrame
> ImportError: cannot import name 'Previews' from 'depthai_sdk' (/home/h/.local/lib/python3.10/site-packages/depthai_sdk/__init__.py)
> Sentry is attempting to send 2 pending error messages
> Waiting up to 2 seconds
> Press Ctrl-C to quit
> Error while running demo script... Command '"/usr/bin/python3" depthai_demo.py --noSupervisor --guiType qt' returned non-zero exit status 1.
> Waiting 5s for the device to be discoverable again...
> Setting up demo...
> Available devices:
> [0] 1844301061D26F0E00 [X_LINK_UNBOOTED]
> USB Connection speed: UsbSpeed.SUPER

themarpe, 2023-02-07:
I tbink the following will fix this issue: https://github.com/luxonis/depthai/pull/933

Cc: @daniilpastukhov 

Thanks all for the report!

closed 2023-02-07 by daniilpastukhov