luxonis/depthai #951: [BUG] {Title of the bug}

dautam, 2023-02-21:
bash -c "$(curl -fL https://docs.luxonis.com/install_dependencies.sh)"
Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 127 at line 164

themarpe, 2023-02-24:
@dautam do you mind posting back the whole log + OS information?

themarpe, 2023-02-24:
CC: @PetrNovota

dautam, 2023-02-25:
Last login: Thu Feb 23 00:34:14 on ttys000
tamdau@Tams-MBP ~ % bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  7635  100  7635    0     0   4881      0  0:00:01  0:00:01 --:--:--  4881
mkdir: /Users/tamdau/Luxonis: File exists

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /Users/tamdau/Luxonis
Directory: /Users/tamdau/Luxonis is OK

Python version: Python 3.11.2 found.
If you want to use it for installation, press ANY key, otherwise input path to python binary.
Press any key to continue

_____________________________
Calling macOS_installer.sh
_____________________________
Running macOS installer.
Installing global dependencies.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  5671  100  5671    0     0   4113      0  0:00:01  0:00:01 --:--:--  4113
During Homebrew install, certain commands need 'sudo'. Requesting access...
Password:

 >> Installing Homebrew from https://raw.githubusercontent.com/Homebrew/install/master/install.sh
Warning: Running in non-interactive mode because `$CI` is set.
==> Checking for `sudo` access (which may request your password)...
==> This script will install:
/opt/homebrew/bin/brew
/opt/homebrew/share/doc/homebrew
/opt/homebrew/share/man/man1/brew.1
/opt/homebrew/share/zsh/site-functions/_brew
/opt/homebrew/etc/bash_completion.d/brew
/opt/homebrew
==> /usr/bin/sudo /usr/sbin/chown -R tamdau:admin /opt/homebrew
==> Downloading and installing Homebrew...
remote: Enumerating objects: 471, done.
remote: Counting objects: 100% (248/248), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 471 (delta 243), reused 245 (delta 243), pack-reused 223
Receiving objects: 100% (471/471), 220.78 KiB | 1.87 MiB/s, done.
Resolving deltas: 100% (305/305), completed with 112 local objects.
From https://github.com/Homebrew/brew
 * [new branch]          dependabot/bundler/Library/Homebrew/concurrent-ruby-1.2.1 -> origin/dependabot/bundler/Library/Homebrew/concurrent-ruby-1.2.1
   336c2c792..0172ad102  master     -> origin/master
HEAD is now at 0172ad102 Merge pull request #14794 from Bo98/api-check-shell
==> Downloading https://formulae.brew.sh/api/formula.json
######################################################################## 100.0%
==> Downloading https://formulae.brew.sh/api/cask.json
######################################################################## 100.0%
Warning: /opt/homebrew/bin is not in your PATH.
  Instructions on how to configure your shell for Homebrew
  can be found in the 'Next steps' section below.
==> Installation successful!

==> Homebrew has enabled anonymous aggregate formulae and cask analytics.
Read the analytics documentation (and how to opt-out) here:
  https://docs.brew.sh/Analytics
No analytics data has been sent yet (nor will any be during this install run).

==> Homebrew is run entirely by unpaid volunteers. Please consider donating:
  https://github.com/Homebrew/brew#donations

==> Next steps:
- Run these two commands in your terminal to add Homebrew to your PATH:
    (echo; echo 'eval "$(/opt/homebrew/bin/brew shellenv)"') >> /Users/tamdau/.zprofile
    eval "$(/opt/homebrew/bin/brew shellenv)"
- Run brew help to get started
- Further documentation:
    https://docs.brew.sh

 >> brew install git
bash: line 93: brew: command not found

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 127 at line 164 

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 127 at line 231 

I try to install a depthai on Mac with this link bash -c "$(curl -fL https://docs.luxonis.com/install_depthai.sh)" and it come up w error bug
I wanna install to test out the luxonis OAK_D camera.

PetrNovota, 2023-02-25:
@dautam 
Thank you for reporting this issue and I am sorry for the inconvenience. I will update the installer.
For now, I suggest a fast solution.

1. open terminal
2. run: `(echo; echo 'eval "$(/opt/homebrew/bin/brew shellenv)"') >> /Users/tamdau/.zprofile`
3. run: `eval "$(/opt/homebrew/bin/brew shellenv)"`
4. run the depthai installer again.

This should help you.
Please let me know whether this worked for you or not.