luxonis/depthai #958: [Feature-Request] Detect computers running macOS as interactive

wvauclain, 2023-02-28:
### Start with the `why`:

Some developers have mac computers and want to view visualizations.

### Move to the `what`:

When running an example like the following on macOS, no windows currently pop up:

```python3
from depthai_sdk import OakCamera

with OakCamera() as oak:
    cam = oak.create_camera('color')
    oak.visualize(cam.out.main)
    oak.start(blocking=True)
```

On the other hand, if you run the program while defining the `DISPLAY` environment variable to a dummy value, it works.

### Move to the `how`:

The current criteria for an interactive computer is linked below:

https://github.com/luxonis/depthai/blob/be73fbc4c8dd355366edf7bc1782387b421eeb83/depthai_sdk/src/depthai_sdk/visualize/visualizer.py#L33

This check needs to be updated to include macOS alongside `nt`. I believe the OS name is `darwin` but would have to double-check that.

themarpe, 2023-03-17:
@wvauclain 

If you have an idea in mind / a fix that handles your case, feel free to submit a PR. Will help bringing this in & fixing

Erol444, 2023-09-01:
1.13 SDK contains refactors for visualizers, so this should be solved.

closed 2023-09-04 by daniilpastukhov