luxonis/depthai #962: Error: Communication exception - possible device error/misconfiguration.

ParthSingh0506, 2023-03-07:
**Describe the bug**
I'm trying to load my custom computer vision TensorFlow model on OAK-D lite. However, I keep encountering the error "Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'preview' (X LINK ERROR)'. I have already tried changing the USB cable from 2.0 to 3.1, but the error persists. 

**Minimal Reproducible Example**
[poseMRE.zip](https://github.com/luxonis/depthai/files/10912159/poseMRE.zip)

**Expected behaviour**
Once the pose_estimation module using the Movenet depthai blob file has successfully identified the key points of a pose, the pre-trained model should be loaded to recognize that pose. The resulting output will display the results of the processing on the OAK-D Lite, indicating that the model has successfully recognized the pose.

_I have used the movenet_singlepose_lightning_U8_transpose.blob file for pose_estimation_

**Attach system log**
[log_system_information_output.txt](https://github.com/luxonis/depthai/files/10911297/log_system_information_output.txt)

Erol444, 2023-03-09:
Hi @ParthSingh0506 ,
I tried to repro this code but I can't:
```
PS C:\Users\Erik\Documents\Work\Luxonis\depthai-experiments> python .\pipe.py
Traceback (most recent call last):
  File "C:\Users\Erik\Documents\Work\Luxonis\depthai-experiments\pipe.py", line 34, in <module>
    pose_recog.setBlobPath('yogapose_ir.blob')
RuntimeError: Cannot load blob, file at path yogapose_ir.blob doesn't exist.
```
Please make sure it's reproducible. And **minimal**. Please also include pipeline graph.
Thanks, Erik

ParthSingh0506, 2023-03-10:
Okay @Erol444, I will make changes and include the pipeline graph.

ParthSingh0506, 2023-03-10:
Hi @Erol444, I will make the code minimal and also include the pipeline graph

ParthSingh0506, 2023-03-10:
Hi @Erol444, I am trying to create a pipeline graph using the depthai library, but I am running into some errors. Specifically, I am having trouble creating a JSON representation of the graph.

I understand that the depthai library provides a method called serializeToJson() that can be used to generate a pipeline graph in JSON format. However, I'm not quite sure how to use this method properly.

Could you please provide some guidance on how to use the serializeToJson() method to create a pipeline graph and then save it as a JSON file?

Erol444, 2023-03-11:
Hi @ParthSingh0506 ,
The [depthai_pipeline_graph](https://github.com/luxonis/depthai_pipeline_graph) tool will abstract this already.
Thanks, Erik

ParthSingh0506, 2023-03-12:
Hi @Erol444, the _depthai_pipeline_graph_ is a library that we have to import into our code? After installing it, when I run the pipeline_graph command it is showing no command _pipeline_graph_ found. I even tried to use the GUI version of the pipeline graph, but in GUI when I tried to give the JSON file it not generating the pipeline graph.

Can you tell me the correct way of generating the JSON file of the pipeline, so that once it is generated we can give it to pipeline_graph, is this the right way of generating the JSON file of the pipeline?

```
device = pipeline.Device()
pipeline_json = pipeline.serializeToJson()

with open("pipeline_config.json", "w") as f:
    json.dump(pipeline_json, f, indent=2)

```

ParthSingh0506, 2023-03-12:
Hi @Erol444 , just wanted to update you that I have simplified the code and uploaded the missing .blob file and uploaded them in poseMRE.zip. However, I'm having trouble generating the pipeline graph. I've uploaded the pipeline in JSON format to the pipeline_graph_json folder. Could you please take a look at the file and see if you can generate the pipeline graph? If possible, could you also let me know the cause of the errors? Thank you for your assistance!"

Erol444, 2023-03-12:
@ParthSingh0506 you can run the utility via `pipeline_graph run "python mre_code.py"`

Erol444, 2023-03-12:
@ParthSingh0506 This isn't even far from the MRE, I have no idea what you are doing, or how you came to the conclusion this is a valid pipeline/program that can run, and that it's an error in depthai:

- Line 8; Should be `dai.Pipeline()`
- Line 28: You are sending NN passhtrough outputs to the NN input? How did you thought this will work?
- Line 37: You are calling `device = pipeline.Device()` - No idea what this is
- Line 47/50 - You are creating new output queues, for which you don't even have XLinkOut nodes? 

Overall, I would strongly suggest:

- Actually reading documentation / demos, and trying them out, instead of just throwing some code together and opening an issue that "It doesn't work!".
- Actually reading READMEs - for eg pipeline graph tool, as from the graph it's clear pipeline isn't valid.
- Use python linting - and linter will tell you that eg. `dai.pipeline()` doesn't work.

Thanks, Erik

closed 2023-03-12 by Erol444