luxonis/depthai #963: Error importing FPS with MovenetDepthaiEdge from depthai_movenet

ParthSingh0506, 2023-03-07:
**Describe the bug**
I am currently working on deploying my custom computer vision TF model on OAK-D. I am attempting to use the pose_estimation feature from depthai_movenet and utilizing the MovenetDepthaiEdge module. However, upon importing the module, I am encountering the error message "no module named FPS". I have already tried reinstalling the package but the issue persists.

**Minimal Reproducible Example**
```
from collections import namedtuple
from pathlib import Path
from FPS import FPS
import depthai as dai
import time

```
**Expected behaviour**
Able to use the MovenetDepthaiEdge class for pose estimation and recognize poses with a custom TensorFlow pose recognition model, making sure that the FPS module is imported without errors from depthai_movenet_. Once the key points have been extracted, the pre-trained model should be able to recognize the pose. The output will then display the detected pose, with the processing taking place on the OAK-D Lite.

**Screenshots**
[image]

**Attach system log**
[log_system_information_output.txt](https://github.com/luxonis/depthai/files/10912261/log_system_information_output.txt)

Erol444, 2023-03-09:
Hi @ParthSingh0506 ,
Could you elaborate which demo are you trying to run? If it's the https://github.com/geaxgx/depthai_movenet, then I'd suggest opening issue there.
Thanks, Erik

ParthSingh0506, 2023-03-10:
Hi @Erol444, while attempting to incorporate the depthai_movenet package into my project, I encountered an issue with the MovenetDepthaiEdge class. Specifically, an error occurred when trying to import the class. I wasn't running any demo files, just my own custom code after importing the class. My intention was to leverage the pose estimation functionality offered by the package in my project.

Erol444, 2023-03-10:
Hi @ParthSingh0506 ,
After digging into it, it seems like you are missing this [FPS file](https://github.com/geaxgx/depthai_movenet/blob/main/FPS.py), which you are trying to import. I am not sure about your application/project structure, but your imports are incorrect - I would recheck those.
Thanks, Erik

ParthSingh0506, 2023-03-10:
@Erol444, I have attempted to install and uninstall the FPS package multiple times. The error is generated only when I try to import the MovenetDepthaiEdge class from the depthai_movenet library in my custom code. Interestingly, when I run their demo file, demo.py, it runs without any errors and no module named FPS is mentioned. The issue seems to only occur when I try to import their library in my custom code.

jakaskerl, 2023-03-10:
Hi @ParthSingh0506,

Please check this guide on how to import python modules from a different directory.
https://www.w3docs.com/snippets/python/importing-files-from-different-folder.html
Have a nice day, Jaka

closed 2023-03-27 by Erol444