luxonis/depthai #967: [BUG] Frame syncing on OAK example does hangs after 3-4 frame pairs

andrebododea, 2023-03-08:
**[x] Check if issue already exists**
- [x] Google it (e.g. *error xy* github luxonis depthai)
- [x] Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
When I use a custom Script node for frame synchronization on-camera as described in [this page](https://docs.luxonis.com/projects/api/en/latest/samples/mixed/frame_sync/), however using the `ColorCamera`'s ISP block as input into the script as opposed to the video video block with is shown in the example, I get the first few packets that come through and time synchronize properly. However after the first 4-5 packets are processed, something in the pipeline hangs and no more packets come into the Script as inputs at all. 

For an A/B comparison, first run my provided example and observe the output which showcases the hanging behavior bug explained above. 

Then simply change line 65 from the current implemented:
```C++
rgb->isp.link(script->inputs["rgb_in"]);
```
to the following, swapping in the video block from the `ColorCamera` node:
```C++
rgb->video.link(script->inputs["rgb_in"]);
```

On my system, the ISP version in the code I provided hangs, but after the modification changing isp to video it works as intended. 

**Minimal Reproducible Example**
> Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)

Attached within the following zip as `main.cpp`: 
[main.zip](https://github.com/luxonis/depthai/files/10925637/main.zip)

**Expected behavior**
> A clear and concise description of what you expected to happen.

**Screenshots**
> If applicable, add screenshots to help explain your problem.

The output of the program is provided here:
[image]

You will notice that the first few frames are synced and make it through the pipeline to the host properly, however after the first few pairs something in the pipeline hangs and there are no more frames provided to the custom Script (see lack of ` [Script(9)] [info] xyz` printouts)

**Pipeline Graph**
> Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).
Not possible via the existing DepthAI pipeline graph tool, as far as I'm aware, due to this particular example being provided as C++ rather than as a Python script.

As far as I'm aware, the pipeline path of interest to this particular issue looks like this:
 (Note: hand-drawn)
[image]

**Attach system log**
 > Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)
Attached as log_system_information.json

**Additional context**
Thinking that there may have been an issue with the timestamp syncing script, I implemented a modified version of the sequence number script provided in `depthai-experiments/gen2-syncing` to run on-camera:

```Python
    script->setScript(R"(
        FPS=30
        import time
        from datetime import timedelta
        import math

        msgs = dict()

        def add_msg(msg, name, seq = None):
            if seq is None:
                seq = msg.getSequenceNum()
            node.info(f"Evaluating seq number for {name}, which is {seq}")
            seq = str(seq)
            if seq not in msgs:
                msgs[seq] = dict()
            msgs[seq][name] = msg

        def get_msgs():
            global msgs
            seq_remove = [] # Arr of sequence numbers to get deleted
            for seq, syncMsgs in msgs.items():
                seq_remove.append(seq) # Will get removed from dict if we find synced msgs pair
                # Check if we have both detections and color frame with this sequence number
                if len(syncMsgs) == 2: # rgb + depth
                    for rm in seq_remove:
                        # node.info(f"Deleting stuff!")
                        node.info(f"Deleting messages with sequence numbers: {msgs[rm].keys()}")
                        del msgs[rm]
                    return syncMsgs # Returned synced msgs
            return None

        names = ['depth', 'rgb']
        frames = dict() # Dict where we store all received frames
        for name in names:
            frames[name] = []

        while True:
            for name in names:
                msg = node.io[name+"_in"].tryGet()
                if msg is not None:
                    # add_msg(msg, name)
                    node.io[name+'_out'].send(msg) # Send all msg frames to the host

            synced = get_msgs()
            if synced:
                node.info(f"Synced frame!")
                frameRgb = synced["rgb"]
                node.info(f"rgb ts: {str(frameRgb.getTimestamp())}, seq {str(frameRgb.getSequenceNum())}")
                frameDepth = synced["depth"]
                node.info(f"depth, ts: {str(frameDepth.getTimestamp())}, seq {str(frameDepth.getSequenceNum())}")
                node.io['rgb_out'].send(frameRgb) # Send synced frames to the host
                node.io['depth_out'].send(frameDepth) # Send synced frames to the host

            time.sleep(0.0001)  # Avoid lazy looping
        )");
```

However - same result as originally described. I get hanging behavior after the first few packets are processed/synced.

Erol444, 2023-03-09:
Hi @andrebododea ,
My guess is that you aren't disposing frames from the script node (instead you are just saving them), which means RGB cam will quickly run out of memory and won't be able to produce new frames (until old ones are disposed). You can try to increase Pool size ([see docs here](https://docs.luxonis.com/projects/api/en/latest/components/nodes/)) of the color cam, eg. `rgb_camera.setNumFramesPool(10,10,10,10,10)` - if this change results in different print, then that's your suspect.
I hope this helps!
Thanks, Erik

andrebododea, 2023-03-15:
Thank you for the super quick reply. This did indeed fix the issue - specifically I needed to use the `setIspNumFramesPool(...)` function, since the ISP block was feeding the custom Script node. Just wanted to mention the ISP-specific version of this function in case anyone in the future comes across this comment with the same root cause issue.

That being said, I think that this silent failure behavior should be mitigated in a much better way, especially because it is a known issue with a known resolution. 

I could see two potential ways to proceed (one more uncertain and one very easy/straightforward):
1. The hard way, but potentially more robust: I wonder if the internal frames pool could be automatically increased if this hanging behavior is encountered. Because I don't know how the internals of the system work, I'm unsure if this is even possible, but there may be a way to catch an insufficiently large frame pool and automatically increase to try to salvage the situation. 
2. The easy way: If the above is too high effort or just not possible, would it be feasible to at least output a warning or error to the console to warn the Depth AI log? That way the user can diagnose this issue directly via log output, rather than needing to be aware of this specific. 

Let me know your thoughts regarding rolling one of the two above items into a feature request. I'm happy to discuss this further as necessary.

closed 2023-03-15 by andrebododea

themarpe, 2023-03-15:
The easy way is being worked on, though is slightly delayed due to some other things jumping in. 
We'll hopefully add this in the upcoming month:)