luxonis/depthai #968: [BUG] Delay using multiple DepthAI cameras in RGB-D mode with multiprocessing

yuriyOn, 2023-03-10:
**Check if issue already exists**

- [x] Google it (e.g. *error xy* github luxonis depthai)
- [x] Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**

There is a significant delay in receiving synchronized RGB-D data that occurs when using OAK-D Pro PoE cameras together, where each camera is handled by a dedicated process. In particular, it occurs after starting the Depth queues of multiple devices from dedicated per-device processes.

I am using multiple OAK-D Pro PoE devices connected to the same host machine. Each camera has a dedicated process using the `multiprocessing` library. Per-camera processes are started from the main process, and then each camera is loaded using the same pipeline in RGB-D mode.

The delay is not present when multiple cameras are in RGB mode, or only one camera's depth queue is initialized (only one camera is in RGB-D mode). It also depends on the cameras that are used together. Some camera combinations do not introduce the delay, while other do.

From extensive tests, I noticed that the issue is introduced when the depth queue is loaded for multiple cameras. Below is a minimal example.

**Steps to Reproduce the Behavior**
- Start the main process, create and start a separate worker process for each camera.
- Use at least two cameras. If the issue cannot be reproduced, try several combinations of cameras and vary the order of camera loading since one camera can have the delay, while another does not. The more cameras are used together, the more likely the issue is to happen.
- In the worker process, initialize the camera with RGB-D pipeline.
- Start the camera queues: Control queue, RGB queue, **Depth queue**
- Capture synchronized RGB-D images by only one camera to avoid influence of data transfer with multiple cameras.
- **Observe delay if depth queue of another camera(s) is initialized**.
- **If the depth queue of the other camera(s) is not initialized, there is no delay**, even if the device is started and Control and RGB queues are initialized.

**Minimal Reproducible Example**

I created [a dedicated repository with the code and ReadMe on usage instructions](https://github.com/yuriyOn/depthai-multiprocessing-issue).

In the minimal reproducible example, I perform the following steps:
- Start the main process, create and start a separate worker process for each camera.
- Use at least two cameras. If the issue cannot be reproduced, try several combinations of cameras and vary the order of camera loading since one camera can have the delay, while another does not. The more cameras are used together, the more likely the issue is to happen.
- In the worker process, initialize the camera with RGB-D pipeline.
- Start the camera queues: Control queue, RGB queue, **Depth queue**
- Capture synchronized RGB-D images by only one camera to avoid influence of data transfer with multiple cameras
- **Observe delay if depth queue of another camera(s) is initialized**
- **If the depth queue of the other camera(s) is not initialized, there is no delay**, even if the device is started and Control and RGB queues are initialized

**Expected behavior**
Starting the depth queue of a camera without data transfer to the host should not affect the data transfer speed of other cameras.

**Screenshots**
Without starting the depth queue of one of the cameras, the other camera consistently provides 12 MP images and synchronized depth maps in approximately 600 ms:

[image]

- Behavior with the delay, **loading depth queue** of the second camera, still not capturing images with the second camera:

[image]

**Additional context**
- The issue can be reproduced consistently using the minimal code sample provided.
- The issue persists even if the frames are not requested from the cameras other than the first one, and yet the delay still occurs when the depth queue of another camera is initialized.
- Experimented with replacing the cable of the camera that has a delay (in combination with first camera) with the one plugged in a camera that does not have a delay (in combination with the first camera), but it did not resolve the issue.
- From the above points, it can be concluded that it is not a network issue, nor it is an issue of the camera hardware or firmware. It appears to be caused by DepthAI using shared memory and not handling device queues separately (during starting the depth queue).
- Introducing a delay to loading the device depth queue sometimes changes the camera that is experiencing the delay.
- The delay does not happen if one camera runs in the RGB-D mode and the second runs in RGB mode.

**Attach system log**
Partial output of the `log_system_information.py` script:
- OS: Ubuntu 22.04.1 LTS
- Network speed: 1 Gbps
- DepthAI Python version: 2.20.2.0
- architecture: 64bit ELF
- processor: x86_64
- platform: Linux-5.19
- python version: 3.9.15
- python build: Nov 24 2022
- python compiler: GCC 11.2.0
- python implementation: CPython

**Pipeline**

The pipeline can be found in `pipeline.py`, it consists of Color camera node and Stereo node, RGB queue, depth queue and control queue.

jakaskerl, 2023-03-12:
Hi @yuriyOn,
I believe the network bandwidth might be saturated. I checked through your sample code and did some rough calculations for the required bandwidth for 12MP RGB camera and 800p depth:

12MP * 30FPS * 1.5 (for NV12) * 2 + 1MP * 10FPS * 2 (depth or 2x mono) = 9Gbit/s

I would suggest lowering the RGB camera resolution first and then lowering the FPS. Alternatively you can choose to encode the frames using H.264, H.265 or MJPEG. Please also check the [docs](https://docs.luxonis.com/projects/api/en/latest/tutorials/low-latency/#poe-latency) for more information regarding low latency on PoE.

Please let me know if this fixes the issue.
Kind regards, 
Jaka

BartvanMarrewijk, 2025-01-24:
Hi @yuriyOn ,

Is it possible to share the code. I experience similar issues and only want to get the data when requested. Thanks in advance. 

Kind regards,

Bart