luxonis/depthai #969: [BUG] Strange Encoder Latency Behaviour

justin-larking-pk, 2023-03-13:
In attempting to figure out why some pipelines have high/increasing latency and in an attempt to produce a low latency streaming pipeline I noticed some strange behaviour from the encoder nodes.

Running the rgb encoder example from the api docs, just with latency measurement, mjpeg codec and at 720p, I get a very respectable 26.5 millisecond average latency:
[image]
(Note a sample is the average latency of 10 frames)

**However**
When running the encoder to stream the rgb, left and right frames latency varies heavily in a repeating pattern. It appears that the decoder and frames are getting out of sync? But then they get so out of sync it resync's back onto itself?
[image]
You can see above this is the same for all codec's. And that the "in sync" latency is essentially the same as the rgb encoding example, with the peak being about double the minimum (roughly 26ms min and 52ms peak).

In addition the behaviour on the mono frames is the reverse:
[image]
It appears to start off desync'ed and improve its latency until it desync's again.

Any idea whats going on here?

justin-larking-pk, 2023-03-15:
I should note the results above were obtained on an oakd-pro-w-poe (ov9782). I decided I should test across other devices.

Here is the same test on an oakd-pro-w-poe(imx378)
rgb latency:
[image]
left latency;
[image]

Same test on an okad-s1:
rgb latency:
[image]
left latency:
[image]

As you can see the oakd-pro-w-poe(imx378) and oakd-s1 dont suffer from this encoder latency desync or whatever is happening. Anyone at luxonis able to shed some light?

justin-larking-pk, 2023-03-15:
Here is the script used to produce latencies:

    import depthai as dai
    import time
    import pickle
    
    # Create pipeline
    pipeline = dai.Pipeline()
    
    pipeline.setXLinkChunkSize(0)
    
    # Define sources and output
    camRgb = pipeline.create(dai.node.ColorCamera)
    videoEnc_rgb = pipeline.create(dai.node.VideoEncoder)
    videoEnc_left = pipeline.create(dai.node.VideoEncoder)
    videoEnc_right = pipeline.create(dai.node.VideoEncoder)
    monoleft = pipeline.create(dai.node.MonoCamera)
    monoright = pipeline.create(dai.node.MonoCamera)
    xout = pipeline.create(dai.node.XLinkOut)
    xout_left = pipeline.create(dai.node.XLinkOut)
    xout_right = pipeline.create(dai.node.XLinkOut)
    
    xout.setStreamName('encoded_rgb')
    xout_left.setStreamName('encoded_left')
    xout_right.setStreamName('encoded_right')
    
    fps=30
    
    # Properties
    camRgb.setBoardSocket(dai.CameraBoardSocket.RGB)
    camRgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_1080_P)
    camRgb.setIspScale(2, 3)
    camRgb.setFps(fps)
    videoEnc_rgb.setDefaultProfilePreset(30, dai.VideoEncoderProperties.Profile.H265_MAIN)
    
    monoleft.setBoardSocket(dai.CameraBoardSocket.LEFT)
    monoleft.setFps(fps)
    monoleft.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    monoright.setBoardSocket(dai.CameraBoardSocket.RIGHT)
    monoright.setFps(fps)
    monoright.setResolution(dai.MonoCameraProperties.SensorResolution.THE_720_P)
    videoEnc_left.setDefaultProfilePreset(fps, dai.VideoEncoderProperties.Profile.H265_MAIN)
    videoEnc_right.setDefaultProfilePreset(fps, dai.VideoEncoderProperties.Profile.H265_MAIN)
    
    camRgb.video.link(videoEnc_rgb.input)
    videoEnc_rgb.bitstream.link(xout.input)
    
    monoleft.out.link(videoEnc_left.input)
    videoEnc_left.bitstream.link(xout_left.input)
    
    monoright.out.link(videoEnc_right.input)
    videoEnc_right.bitstream.link(xout_right.input)
    
    DEVICE = "OAKD"
    
    with dai.Device(pipeline) as device:
        #device.setLogLevel(dai.LogLevel.DEBUG)
        #device.setLogOutputLevel(dai.LogLevel.DEBUG)
    
        q_rgb = device.getOutputQueue(name="encoded_rgb", maxSize=1, blocking=False)
        q_left = device.getOutputQueue(name="encoded_left", maxSize=1, blocking=False)
        q_right = device.getOutputQueue(name="encoded_right", maxSize=1, blocking=False)
        latencies = []
        rgb_sum = 0
        left_sum = 0
        right_sum = 0
        count = 0
        while count<10001:
            Packet_rgb = q_rgb.get()  # Blocking call, will wait until a new data has arrived
            rgb_latency = (dai.Clock.now() - Packet_rgb.getTimestamp()).total_seconds()
            rgb_sum+=rgb_latency
            print("RGB Latency(ms):"+str(rgb_latency*1000))
            Packet_left = q_left.get()  # Blocking call, will wait until a new data has arrived
            left_latency = (dai.Clock.now() - Packet_left.getTimestamp()).total_seconds()
            left_sum+=left_latency
            print("Left Latency(ms):"+str(left_latency*1000))
            Packet_right = q_right.get()  # Blocking call, will wait until a new data has arrived
            right_latency = (dai.Clock.now() - Packet_right.getTimestamp()).total_seconds()
            right_sum+=right_latency
            print("Right Latency(ms):"+str(right_latency*1000))
            if(count % 10 == 0):
                cpuusage = device.getLeonCssCpuUsage()
                latencies.append([rgb_sum/10, left_sum/10, right_sum/10, cpuusage.average])
                rgb_sum=0
                left_sum=0
                right_sum=0
            count+=1
    
        pickle.dump(latencies, open("OAKD_OB9782_H265.p", "wb"))

justin-larking-pk, 2023-03-21:
Is anyone at luxonis able to recreate?

themarpe, 2023-03-21:
Hi @justin-larking-pk 

Sorry for the delay - due to current load, I don't think we'll be able to dive deep into this one, but one thing to perhaps check is the pool sizes of all cameras in use. Try increasing those and see if that helps mitigate the issue.

justin-larking-pk, 2023-03-22:
This revealed more severe behaviour. On the oakd-pro-w-poe(ov9782 variant) using that script and setting queue size to 30 (i.e. maxSize=30) the latency for the rgb stream just grew over time, ending at a whopping 250ms from 90ms at the start. Interestingly the left view stayed at around 30ms the whole time.
[image]

Running the same script with a queue of 30 on the oakd-pro-w-poe-imx378 showed no latency buildup:
[image]