luxonis/depthai #971: [BUG] Red/pink hues in RGB images produced by the OV9782 sensor on part no. OAK-D-Pro-W-97

andrebododea, 2023-03-15:
**Check if issue already exists**
[x] Google it (e.g. *error xy* github luxonis depthai)
[x] Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
This issue is captured in great detail here: https://www.luxonis.com/blog/lens_color_filtering_enhancement

I am experiencing much the same issue, however this does not only appear in outdoor environments - in fact I have only been testing in an indoor office environment (e.g. mainly under fluorescent lighting).

Here is the image I'm seeing (plain whiteboard being used to highlight issue):
[image]

**Minimal Reproducible Example**
Very straightforward MRE (single .cpp file):
[main.zip](https://github.com/luxonis/depthai/files/10983875/main.zip)

After reading [this](https://www.luxonis.com/blog/lens_color_filtering_enhancement) blog post on the Lense Shading Correction issue, and also [this page on Camera tuning for the OV9782](https://docs.luxonis.com/projects/api/en/latest/tutorials/image_quality/#color-camera-isp-configuration), I went ahead and downloaded the [indicated tuning blob for the OV9782 sensor](https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/misc/tuning_color_ov9782_wide_fov.bin) and dropped the following in at line 9 of the above .cpp file:
```C++
int main() {
     // Create pipeline
     dai::Pipeline pipeline;
     pipeline.setCameraTuningBlobPath("/home/andre/Downloads/tuning_color_ov9782_wide_fov.bin");
```
and when I re-run the program this is the output image:
[image]

As you can see, there is still a significant and unacceptable amount of red/pink hue affecting the image.

**Expected behavior**
I would have expected the new camera tuning blob to help the red/pink hue issue, even if at least marginally. See above for more info.

justin-larking-pk, 2023-03-16:
I haven't received a replacement yet. You can check to see if this is actually the issue by getting an IR LED and seeing if the camera can see the IR LED in an else dark environment. If it has the IR filter like the imx378 type cameras you should only see a little red dot. If you see lots of red then the cameras have the issue.

cafemoloko, 2023-03-22:
Thank you for reporting. We will replace the faulty devices - please reach out to us (support@luxonis.com) if you haven't already to discuss the next steps. 

The new batch of units should be available any day now. We are sorry for the issues and delays.

Thank you,
Karolina