luxonis/depthai #973: Not able to pip install depthai-sdk

franva, 2023-03-17:


franva, 2023-03-19:
Btw, I have 
opencv-python          4.4.0.46
opencv-python-headless 4.1.1.26
installed.

So if the depthAI-sdk doesn't require all opencv functions, maybe a quick fix could be replace the `opencv-contrib-python` with `opencv-python`.

themarpe, 2023-03-19:
@franva 

You must only have one opencv installed. As per: https://github.com/opencv/opencv-python#installation-and-usage

I suggest removing all others & just retain the contrib. 
Contrib is needed due to bringing in the required parts for calibration, etc...

themarpe, 2023-03-19:
CC: @daniilpastukhov - if we remove other opencv version in `install_requirements.py` make sure we also uninstall variants like `-headless`, etc...

franva, 2023-03-20:
> @franva
> 
> You must only have one opencv installed. As per: https://github.com/opencv/opencv-python#installation-and-usage
> 
> I suggest removing all others & just retain the contrib. Contrib is needed due to bringing in the required parts for calibration, etc...

Hi @themarpe 
Thanks for the suggestion.

I am a bit hesitate to uninstall opencv-python as I remember it took me a bit while to install it last year(errored out so many times so many places etc.).

Did you say that the issue is possibly caused by the existing opencv-python lib in my RPi system?
If yes for sure, then I can try to uninstall my opencv-python for installing the full package of opencv.

Thanks,

Winston

themarpe, 2023-03-20:
> Did you say that the issue is possibly caused by the existing opencv-python lib in my RPi system?
If yes for sure, then I can try to uninstall my opencv-python for installing the full package of opencv.

It might be. According to their doc, there shouldn't be multiples of those.
Backup your system or create a copy of the lib - I cannot be for sure on this one unfortunatelly