luxonis/depthai #979: [BUG] High Latency while using OAK-D-W

veritas-Qiu, 2023-03-30:
**Check if issue already exists**

reference docs for Latency: https://docs.luxonis.com/projects/api/tutorials/low-latency.html#low-latency

**Describe the bug**
very high latency when using OAK-D-W
different scenes:

1. one mono on, latency is as low as the reference page
2. two monos on @400P60fps, latency is as low as the reference page
3. two monos on @400P120fps, get raw10 output, latency is as low as the reference page
4. two monos on @400P120fps, get grayscale output, latency is about 10x larger than the reference page
5. RGB on @ 1080P30, latency is as low as the reference page
6. RGB on @ 1080P60, latency is about 2x-3x larger than the reference page
7. two monos on @400P120fps, RGB on @ 1080P60, latency is quite high

common ideas like block/unblock queue have been tried, but nothing changes.

**Minimal Reproducible Example**
```
int main() {
    // Create pipeline
    dai::Pipeline pipeline;
    pipeline.setXLinkChunkSize(0);

    // Define sources and outputs
    auto camRgb = pipeline.create<dai::node::ColorCamera>();
    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();
    auto xoutLeft = pipeline.create<dai::node::XLinkOut>();
    auto xoutColor = pipeline.create<dai::node::XLinkOut>();
    auto xoutRight = pipeline.create<dai::node::XLinkOut>();

    xoutLeft->setStreamName("left");
    xoutRight->setStreamName("right");
    xoutColor->setStreamName("color");

    // Properties
    monoLeft->setBoardSocket(dai::CameraBoardSocket::LEFT);
    monoLeft->setResolution(dai::MonoCameraProperties::SensorResolution::THE_400_P);
    monoLeft->setFps(120);
    monoRight->setBoardSocket(dai::CameraBoardSocket::RIGHT);
    monoRight->setResolution(dai::MonoCameraProperties::SensorResolution::THE_400_P);
    monoRight->setFps(120);
    camRgb->setBoardSocket(dai::CameraBoardSocket::RGB);
    camRgb->setResolution(dai::ColorCameraProperties::SensorResolution::THE_1080_P);
    camRgb->setFps(60);

    // Linking
    monoRight->raw.link(xoutRight->input);
    monoLeft->raw.link(xoutLeft->input);
    camRgb->isp.link(xoutColor->input);

    // Connect to device and start pipeline
    dai::Device device(pipeline);

    // Output queues will be used to get the grayscale frames from the outputs defined above
    // auto qLeft = device.getOutputQueue("left", 1, false);
    // auto qRight = device.getOutputQueue("right", 1, false);
    auto qColor = device.getOutputQueue("color", 1, false);

    while(true) {
        // Instead of get (blocking), we use tryGet (non-blocking) which will return the available data or None otherwise
        // auto inLeft = qLeft->get<dai::ImgFrame>();
        // auto inRight = qRight->get<dai::ImgFrame>();
        auto inColor = qColor->get<dai::ImgFrame>();

        if (inColor)
        {
            auto latencyc = (std::chrono::steady_clock::now() - inColor->getTimestamp()).count()/1'000'000;
            std::cout << "latencyc = " << latencyc << std::endl;
        }
    }
    return 0;
}
```

**Expected behavior**
print the latency as low as the reference

**Screenshots**
print the latency higher than the reference

**Pipeline Graph**

**Attach system log**

**Additional context**

Erol444, 2023-03-30:
Hi @veritas-Qiu , is this USB3?

veritas-Qiu, 2023-03-31:
> Hi @veritas-Qiu , is this USB3?

Hi @Erol444 , a C to C cable is used, it should not be USB2. As double-checked, the frame rate of rgb 1080P60 mode is actually 60fps, which perhaps means the bandwidth is higher than 1Gbps and USB2 could not achieve.