luxonis/depthai #981: [BUG] {connectivity issue, or perhaps invalid pipeline}

fisakhan, 2023-03-31:
**Check if issue already exists**

- Google it (e.g. *error xy* github luxonis depthai)
- Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
A clear and concise description of what the bug is.
Running [this](https://docs.luxonis.com/projects/api/en/latest/samples/StereoDepth/rgb_depth_aligned/#rgb-depth-alignment) example, I get the following error:

> [2023-03-31 18:20:25.772] [warning] Monitor thread (device: 1944301021D1741300 [192.168.178.180]) - ping was missed, closing the device connection
> Traceback (most recent call last):
> File "/oak-d/rgb-depth-align-12mp.py", line 101, in <module>
> queueEvents = device.getQueueEvents(("rgb", "disp"))
> ValueError: Device already closed or disconnected

**Minimal Reproducible Example**
Append the MRE to the bug report, [instructions here](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example)
Camera: OAK-D PRO POE
depthai==2.17.3.0
OS: Ubuntu 22.04.1 LTS

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**

Please also provide a screenshot of your pipeline using the [DepthAI Pipeline Graph](https://github.com/geaxgx/depthai_pipeline_graph).

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

deworben, 2024-01-29:
I am having the same issue! I'm trying to connect my OAK-1 depthai camera to my Windows computer with the Hello World program. It works some of the time but I usually have to run the program many times  (about 20 times) to get it to work otherwise it will error out. I was using an old sdk and then upgraded the sdk which changed the error but not the behaviour (it still can't connect).

Error old sdk:
```
Traceback (most recent call last):
  File "C:\Users\bdewo\examples\ColorCamera\rgb_preview.py", line 24, in <module>
    with dai.Device(pipeline) as device:
RuntimeError: Device already closed or disconnected: io error
```

Error new sdk:
```
[1844301091D6321300] [1.2] [1706496136.940] [host] [warning] Device crashed, but no crash dump could be extracted.
Stack trace (most recent call last):
#31   Object "", at 00007FF837F20860, in PyRun_AnyFileExFlags
#30   Object "", at 00007FF837F1EBF3, in PyRun_SimpleFileExFlags
#29   Object "", at 00007FF837F1E11C, in PyImport_GetModuleDict
#28   Object "", at 00007FF837F1E4AF, in Py_stat
#27   Object "", at 00007FF837E8B09A, in PyFuture_FromASTObject
#26   Object "", at 00007FF837E8B11A, in PyFuture_FromASTObject
#25   Object "", at 00007FF837E8B3F9, in PyEval_EvalCode
#24   Object "", at 00007FF837E8B49B, in PyEval_EvalCodeEx
#23   Object "", at 00007FF837E9B815, in PyEval_EvalCodeWithName
#22   Object "", at 00007FF837ED95A3, in PyDict_GetItem
#21   Object "", at 00007FF837EE0735, in PyEval_EvalFrameDefault
#20   Object "", at 00007FF852CC7386, in pybind11::error_already_set::error_already_set  
#19   Object "", at 00007FF837EAC82D, in PyType_GenericNew
#18   Object "", at 00007FF837EAE84C, in PyArg_ParseTuple_SizeT
#17   Object "", at 00007FF837EA4EA7, in PyObject_Call
#16   Object "", at 00007FF837EA50B8, in PyVectorcall_Call
#15   Object "", at 00007FF837FFA034, in Py_gitversion
#14   Object "", at 00007FF837ED884C, in PyObject_MakeTpCall
#13   Object "", at 00007FF837EB051A, in PyDict_MergeFromSeq2
#12   Object "", at 00007FF852CCAC8E, in pybind11::error_already_set::error_already_set  
#11   Object "", at 00007FF852D49254, in PyInit_depthai
#10   Object "", at 00007FF852D4917C, in PyInit_depthai
#9    Object "", at 00007FF852F5659C, in PyInit_depthai
#8    Object "", at 00007FF852F2FC82, in PyInit_depthai
#7    Object "", at 00007FF852F48810, in PyInit_depthai
#6    Object "", at 00007FF8C3153C66, in RtlCaptureContext2
#5    Object "", at 00007FF853058D91, in PyInit_depthai
#4    Object "", at 00007FF853061D20, in PyInit_depthai
#3    Object "", at 00007FF853097C16, in PyInit_depthai
#2    Object "", at 00007FF852F35CAB, in PyInit_depthai
#1    Object "", at 00007FF852F36A7B, in PyInit_depthai
#0    Object "", at 00007FF852F31EED, in PyInit_depthai
```

Erol444, 2024-01-29:
Hi @deworben ,
Have you already tried with a powered USB hub or [OAK Y-adapter](https://docs.luxonis.com/projects/hardware/en/latest/pages/DM6010/), so power doesn't come from your Windows machine? It might be the OAK device brownsout as your computer can't provide enough power.

If you tried that already, please also provide:
- Pip versions of opencv and depthai
- [debug terminal output ](https://docs.luxonis.com/projects/api/en/latest/tutorials/debugging/) from depthai

deworben, 2024-01-30:
Hi @Erol444 you're a genius thanks yep my laptop wasn't able to deliver enough current as it turns out. A passthrough USB hub worked a charm. Thank you!

Erol444, 2024-01-30:
Great, I'm glad it worked!

closed 2024-01-30 by Erol444