luxonis/depthai #986: [SDK Feature-Request] rerun as (default) visualizer for oak.visualize()

Erol444, 2023-04-03:
### Start with the `why`:
OpenCV approach for visualization is quite limited - it doesn't easily support graphs (eg. IMU logging) or pointclouds (would need open3d) and has some strange quirks (example: [cv2.waitKey(1) takes 15ms on Windows](https://github.com/luxonis/depthai/issues/985#issuecomment-1494728216), `cv2.imshow()` has to be run in main thread...).

[Rerun](https://www.rerun.io/) is a new visualization SDK/program that is fast and offer many more options.

### Move to the `what`:
At `oak.visualize()`, have another parameter specifying visualizer. Add `rerun` as an option (along cv2). If rerun, create viewer and instead of cv2.imshow, log frames/imus/pointclouds (when supported) with rerun python SDK.

Here's [initial demo for rerun + depthai](https://github.com/luxonis/depthai-experiments/tree/master/gen2-rerun#gen2-rerun).

### Move to the `how`:
To support multiple visualizers, we would likely need to change the SDK architecture of the codebase a bit, eg. abstract visualizer which cv2/rerun visualizers would build upon, and that would have function to visualize frame / imu msgs / pointclouds / etc.

brmarkus, 2023-04-04:
@Erol444 do you already know and have decided (and checked with the authors of "Rerun") under which license you will make this new dependency available to the users? (They seem to still work on their commercial license, according to "https://www.rerun.io/pricing")

Erol444, 2023-04-04:
Hi @brmarkus ,
We are targeting the OSS project ([see here](https://github.com/rerun-io/rerun#business-model)) via either MIT/Apache, and will (potentially) share any changes back to their main repo. From what I understand, the commercial license will be for cross-team collaboration, which we will not target, but our users might still be able to use those. Thoughts?
Thanks, Erik

brmarkus, 2023-04-04:
Sounds good, thanks for the feedback.

HopeCollector, 2024-03-22:
I found that when using both `oak.record` and `oak.visualize`, the main thread gets stuck in `cv2.imshow`