luxonis/depthai #991: Install Error When Running install_depthai.sh

cwillboyd, 2023-04-09:
I am trying to install depthai on my ubuntu 20.04 machine. I am getting an error stating "/usr/local/bin/python3: No module named pip" despite the fact that I have pip installed. Here is the output of the terminal, can someone help me resolve this issue ? 

sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /home/christopher/Luxonis
Directory: /home/christopher/Luxonis is OK

Python version: Python 3.11.1 found.
If you want to use it for installation, press ANY key, otherwise input path to python binary.
Press any key to continue
_____________________________
Calling linux_installer.sh
_____________________________
Updating sudo-apt.
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://dl.google.com/linux/chrome/deb stable InRelease                               
Hit:3 http://security.ubuntu.com/ubuntu focal-security InRelease                           
Hit:4 http://ca.archive.ubuntu.com/ubuntu focal InRelease                                  
Hit:5 http://ca.archive.ubuntu.com/ubuntu focal-updates InRelease   
Hit:6 http://ca.archive.ubuntu.com/ubuntu focal-backports InRelease 
Hit:7 http://ppa.launchpad.net/appimagelauncher-team/stable/ubuntu focal InRelease
Hit:8 http://repository.spotify.com stable InRelease
Hit:9 http://ppa.launchpad.net/gnome-terminator/ppa/ubuntu focal InRelease
Hit:10 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu focal InRelease
Hit:11 http://ppa.launchpad.net/solaar-unifying/ppa/ubuntu focal InRelease
Reading package lists... Done
Installing global dependencies.
Hit:1 http://ca.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://ca.archive.ubuntu.com/ubuntu focal-updates InRelease                          
Hit:3 http://ca.archive.ubuntu.com/ubuntu focal-backports InRelease                        
Hit:4 http://packages.microsoft.com/repos/code stable InRelease                            
Hit:5 http://dl.google.com/linux/chrome/deb stable InRelease                               
Hit:6 http://ppa.launchpad.net/appimagelauncher-team/stable/ubuntu focal InRelease        
Hit:7 http://security.ubuntu.com/ubuntu focal-security InRelease    
Hit:8 http://repository.spotify.com stable InRelease                       
Hit:9 http://ppa.launchpad.net/gnome-terminator/ppa/ubuntu focal InRelease 
Hit:10 http://ppa.launchpad.net/obsproject/obs-studio/ubuntu focal InRelease
Hit:11 http://ppa.launchpad.net/solaar-unifying/ppa/ubuntu focal InRelease
Reading package lists... Done                        
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libjpeg-dev is already the newest version (8c-2ubuntu8).
libpng-dev is already the newest version (1.6.37-2).
libsm6 is already the newest version (2:1.2.3-1).
libxext6 is already the newest version (2:1.3.4-0ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu4).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-dev is already the newest version (3.8.2-0ubuntu2).
libdc1394-22-dev is already the newest version (2.2.5-2.1).
libgtk2.0-dev is already the newest version (2.24.32-4ubuntu4).
libtbb-dev is already the newest version (2020.1-2).
libtbb2 is already the newest version (2020.1-2).
python3-pyqt5 is already the newest version (5.14.1+dfsg-3build1).
python3-pyqt5.qtquick is already the newest version (5.14.1+dfsg-3build1).
qml-module-qt-labs-platform is already the newest version (5.12.8+dfsg-0ubuntu1).
qml-module-qtquick-controls2 is already the newest version (5.12.8+dfsg-0ubuntu1).
qml-module-qtquick-layouts is already the newest version (5.12.8-0ubuntu1).
qml-module-qtquick-window2 is already the newest version (5.12.8-0ubuntu1).
qml-module-qtquick2 is already the newest version (5.12.8-0ubuntu1).
qtchooser is already the newest version (66-2build1).
qtdeclarative5-dev is already the newest version (5.12.8-0ubuntu1).
build-essential is already the newest version (12.8ubuntu1.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
git is already the newest version (1:2.25.1-1ubuntu3.10).
libgl1-mesa-glx is already the newest version (21.2.6-0ubuntu0.1~20.04.2).
libtiff-dev is already the newest version (4.1.0+git191117-2ubuntu0.20.04.8).
python3-numpy is already the newest version (1:1.17.4-5ubuntu3.1).
udev is already the newest version (245.4-4ubuntu3.20).
ffmpeg is already the newest version (7:4.2.7-0ubuntu0.1).
libavcodec-dev is already the newest version (7:4.2.7-0ubuntu0.1).
libavformat-dev is already the newest version (7:4.2.7-0ubuntu0.1).
libswscale-dev is already the newest version (7:4.2.7-0ubuntu0.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.8).
qt5-qmake is already the newest version (5.12.8+dfsg-0ubuntu2.1).
qtbase5-dev is already the newest version (5.12.8+dfsg-0ubuntu2.1).
qtbase5-dev-tools is already the newest version (5.12.8+dfsg-0ubuntu2.1).
0 upgraded, 0 newly installed, 0 to remove and 67 not upgraded.
/usr/local/bin/python3: No module named pip

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 120 

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 1 at line 192

themarpe, 2023-04-10:
Thanks for the report @cwillboyd 

CC: @PetrNovota it detected Python3.11 while Ubuntu20.04 defaults to 3.8 (which the python3-pip installs pip for as well). Also an `python3 -m ensurepip` could be issued beforehand if need may be.

PetrNovota, 2023-04-10:
@cwillboyd thank you for the report. If you try running the `python3 -m ensurepip` command prior to running the installer, does it help?