luxonis/depthai #992: I have trained the model using yolov8 for pedestrian, cyclist and skateboards. Same deployed on Luxonis. But unable to predict on new image with less brightness or night time. Could u help me? any suggestions?

drakshayanidesai, 2023-04-13:
### Start with the `why`:
Describe why this feature would be useful to you and to other users.

### Move to the `what`:
A clear and concise description of the feature you would like to get added to the `depthai`.

### Move to the `how`:
If you already have an idea of how this new feature could be implemented, describe it here.

brmarkus, 2023-04-14:
Can you provide more details, please?
Which camera-model(s) do you use?
Which input-images have you used?
Can you give examples of images which won't work well - images you used and results you got?
Are the reduced brightness- or night-time-conditions conditions you have to fulfill? Then you better include such conditions in your train- and validation-test-sets. You could use image-editors via scripts to modify existing and "generate" new images and retain the POIs/bounding-box-labels for training.
Do your used camera-models have capabilities for infrared, auto-/manual-white-balancing, contrast-/brightness-settings ("PROC-amp" parameters)?

drakshayanidesai, 2023-04-14:
Hi 
I am using luxonis OKA-D pro

Thanks
On Fri, 14 Apr 2023 13:44:50 +0530 Markus Broghammer  wrote

Can you provide more details, please?

Which camera-model(s) do you use?

Which input-images have you used?

Can you give examples of images which won't work well - images you used and results you got?

Are the reduced brightness- or night-time-conditions conditions you have to fulfill? Then you better 
include such conditions in your train- and validation-test-sets. You could use image-editors via 
scripts to modify existing and "generate" new images and retain the POIs/bounding-box-labels for 
training.

Do your used camera-models have capabilities for infrared, auto-/manual-white-balancing, 
contrast-/brightness-settings ("PROC-amp" parameters)?

Erol444, 2023-06-15:
Invalid bug report.

closed 2023-06-15 by Erol444