luxonis/depthai #996: instlling depthai on jetson agx orin?

DDopfer, 2023-04-16:
apologies, if I'm posting a repeating question, but I would like to run OAK1 cameras on Jetson AGX Orin and installing depthai depndencies results in theses errors, could you advise?:

sudo wget -qO- https://docs.luxonis.com/install_depthai.sh | bash
[sudo] password for oreo1: 
mkdir: cannot create directory ‘/home/oreo1/Luxonis’: File exists

ENTER absolute installation path for depthai or leave empty and default path: $HOME will be used.

Using default installation path: /home/oreo1/Luxonis
Directory: /home/oreo1/Luxonis is OK

Python version: Python 3.8.10 found.
If you want to use it for installation, press ANY key, otherwise input path to python binary.
Press any key to continue
_____________________________
Calling linux_installer.sh
_____________________________
Updating sudo-apt.
Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:2 https://repo.download.nvidia.com/jetson/common r35.3 InRelease
Get:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Hit:4 https://repo.download.nvidia.com/jetson/t234 r35.3 InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Fetched 114 kB in 1s (87.6 kB/s)
Reading package lists... Done
Installing global dependencies.
Hit:1 https://repo.download.nvidia.com/jetson/common r35.3 InRelease
Hit:2 https://repo.download.nvidia.com/jetson/t234 r35.3 InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libjpeg-dev is already the newest version (8c-2ubuntu8).
libpng-dev is already the newest version (1.6.37-2).
libsm6 is already the newest version (2:1.2.3-1).
libxext6 is already the newest version (2:1.3.4-0ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu4).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-dev is already the newest version (3.8.2-0ubuntu2).
libdc1394-22-dev is already the newest version (2.2.5-2.1).
libgtk2.0-dev is already the newest version (2.24.32-4ubuntu4).
libtbb-dev is already the newest version (2020.1-2).
libtbb2 is already the newest version (2020.1-2).
python3-pyqt5 is already the newest version (5.14.1+dfsg-3build1).
python3-pyqt5.qtquick is already the newest version (5.14.1+dfsg-3build1).
qml-module-qt-labs-platform is already the newest version (5.12.8+dfsg-0ubuntu1).
qml-module-qtquick-controls2 is already the newest version (5.12.8+dfsg-0ubuntu1).
qml-module-qtquick-layouts is already the newest version (5.12.8-0ubuntu1).
qml-module-qtquick-window2 is already the newest version (5.12.8-0ubuntu1).
qml-module-qtquick2 is already the newest version (5.12.8-0ubuntu1).
qtchooser is already the newest version (66-2build1).
qtdeclarative5-dev is already the newest version (5.12.8-0ubuntu1).
build-essential is already the newest version (12.8ubuntu1.1).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
git is already the newest version (1:2.25.1-1ubuntu3.10).
libgl1-mesa-glx is already the newest version (21.2.6-0ubuntu0.1~20.04.2).
libtiff-dev is already the newest version (4.1.0+git191117-2ubuntu0.20.04.8).
python3-numpy is already the newest version (1:1.17.4-5ubuntu3.1).
udev is already the newest version (245.4-4ubuntu3.21).
ffmpeg is already the newest version (7:4.2.7-0ubuntu0.1).
libavcodec-dev is already the newest version (7:4.2.7-0ubuntu0.1).
libavformat-dev is already the newest version (7:4.2.7-0ubuntu0.1).
libswscale-dev is already the newest version (7:4.2.7-0ubuntu0.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.8).
qt5-qmake is already the newest version (5.12.8+dfsg-0ubuntu2.1).
qtbase5-dev is already the newest version (5.12.8+dfsg-0ubuntu2.1).
qtbase5-dev-tools is already the newest version (5.12.8+dfsg-0ubuntu2.1).
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dctrl-tools dpkg-repack gdal-data gir1.2-timezonemap-1.0 gir1.2-xkl-1.0
  grub-common libaec0 libarmadillo9 libarpack2 libavresample-dev libcfitsio8
  libcharls2 libdap25 libdapclient6v5 libdebian-installer4 libepsilon1
  libexif-dev libfreexl1 libfwupdplugin1 libfyba0 libgdal26 libgdcm-dev
  libgdcm3.0 libgeos-3.8.0 libgeos-c1v5 libgeotiff5 libgl2ps1.4 libgphoto2-dev
  libhdf4-0-alt libhdf5-103 libhdf5-openmpi-103 libilmbase-dev libkmlbase1
  libkmldom1 libkmlengine1 liblept5 libminizip1 libnetcdf-c++4 libnetcdf15
  libodbc1 libogdi4.1 libopencv-calib3d4.2 libopencv-contrib4.2
  libopencv-dnn4.2 libopencv-features2d4.2 libopencv-flann4.2
  libopencv-highgui4.2 libopencv-imgcodecs4.2 libopencv-imgproc4.2
  libopencv-ml4.2 libopencv-objdetect4.2 libopencv-photo4.2 libopencv-shape4.2
  libopencv-stitching4.2 libopencv-superres4.2 libopencv-video4.2
  libopencv-videoio4.2 libopencv-videostab4.2 libopencv-viz4.2
  libopencv4.2-java libopencv4.2-jni libopenexr-dev libpq5 libproj15 libqhull7
  libsocket++1 libspatialite7 libsuperlu5 libsz2 libtesseract4
  libtimezonemap-data libtimezonemap1 liburiparser1 libvtk6.3 libxerces-c3.2
  libxmlb1 odbcinst odbcinst1debian2 os-prober proj-data python3-icu
  python3-pam rdate tasksel tasksel-data ubuntu-advantage-desktop-daemon
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
Requirement already satisfied: pip in ./oak1/lib/python3.8/site-packages (23.1)
Installing udev rules...
Finished installing global libraries.

Running Linux installer.
Demo app already downloaded. Checking out main and updating.
Already on 'main'
Your branch is up to date with 'origin/main'.
Already up to date.
Creating python virtual environment in /home/oreo1/Luxonis/venv
Requirement already satisfied: pip in /home/oreo1/Luxonis/venv/lib/python3.8/site-packages (23.1)
Requirement already satisfied: packaging in /home/oreo1/Luxonis/venv/lib/python3.8/site-packages (23.1)
Collecting pyqt5
  Using cached PyQt5-5.15.9.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... -bash: line 232: 18877 Killed                  pip install pyqt5

Failed installing dependencies. Could be a bug in the installer or unsupported platform. Open a bug report over at https://github.com/luxonis/depthai - exited with status 137 at line 228

themarpe, 2023-04-18:
CC: @PetrNovota

PetrNovota, 2023-04-18:
@DDopfer I believe your issue is the same as [this one](https://github.com/luxonis/depthai/issues/897#issuecomment-1413750667) The problem you are facing is the PyQt5 installation, it should be installed globally via
`sudo apt install python3-pyqt5`
The installation script failed on the last step so if you manage to install the pyqt5 via the above command, then you should be all set.
Then it should be enough source the virtual environment from `/home/oreo1/Luxonis/venv/bin` and start the demo `depthai_demo.py`
Please, let me know if it works

PetrNovota, 2023-04-18:
@DDopfer we are sorry for the inconvenience and thanks for confirming that it worked. I will update the installation script

closed 2023-04-18 by PetrNovota

DDopfer, 2023-04-18:
Petr,Please don't worry about it. This was painless compared to other problems that needed to be solved.Thank you for your help.I will start trying to run custom object detection models on the AGX Orin next, where we read cows' eartags for identification in the front of the cow and detect disease in the back of the cow using two cameras.Models run on a laptop w OAK1PoE cameras (for water resistance), but I want to use a portable microcomputer .... and see how fast the models can run in real time to sort cows for treatment after milking.I want to apply this on a large (!) dairy farm starting in May.Have a peaceful day.Dörte D Döpfer +1 (608) 698 7812

PetrNovota, 2023-04-18:
@DDopfer thank you. Wish you good like with your project. If you have any troubles, let us know

DDopfer, 2023-04-19:
Petr
sorry for the bother, but now I am trying to install the depthai
dependencies on a fresh install of ubuntu 22.04 on a Lenovo Yoga laptop and
I get into trouble again (I have installed pytq5 first):
(oak1) ***@***.***:~$ sudo wget -qO-
https://docs.luxonis.com/install_dependencies.sh | bash
Hit:1 http://us.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:3 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:4 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
libjpeg-dev is already the newest version (8c-2ubuntu10).
libpng-dev is already the newest version (1.6.37-3build5).
libsm6 is already the newest version (2:1.2.3-1build2).
libxext6 is already the newest version (2:1.3.4-1build1).
pkg-config is already the newest version (0.29.2-1ubuntu3).
libdc1394-dev is already the newest version (2.2.6-4).
libgtk2.0-dev is already the newest version (2.24.33-2ubuntu2).
libtbb-dev is already the newest version (2021.5.0-7ubuntu2).
libtbb2 is already the newest version (2020.3-1ubuntu3).
python3-pyqt5 is already the newest version (5.15.6+dfsg-1ubuntu3).
python3-pyqt5.qtquick is already the newest version (5.15.6+dfsg-1ubuntu3).
qml-module-qt-labs-platform is already the newest version (5.15.3+dfsg-1).
qml-module-qtquick-controls2 is already the newest version (5.15.3+dfsg-1).
qml-module-qtquick-layouts is already the newest version (5.15.3+dfsg-1).
qml-module-qtquick-window2 is already the newest version (5.15.3+dfsg-1).
qml-module-qtquick2 is already the newest version (5.15.3+dfsg-1).
qtchooser is already the newest version (66-2build1).
qtdeclarative5-dev is already the newest version (5.15.3+dfsg-1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
git is already the newest version (1:2.34.1-1ubuntu1.8).
libtiff-dev is already the newest version (4.3.0-6ubuntu0.4).
python3 is already the newest version (3.10.6-1~22.04).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-numpy is already the newest version (1:1.21.5-1ubuntu22.04.1).
udev is already the newest version (249.11-0ubuntu3.9).
ffmpeg is already the newest version (7:4.4.2-0ubuntu0.22.04.1).
libavcodec-dev is already the newest version (7:4.4.2-0ubuntu0.22.04.1).
libavformat-dev is already the newest version (7:4.4.2-0ubuntu0.22.04.1).
libgl1-mesa-glx is already the newest version (22.2.5-0ubuntu0.1~22.04.1).
libswscale-dev is already the newest version (7:4.4.2-0ubuntu0.22.04.1).
qt5-qmake is already the newest version (5.15.3+dfsg-2ubuntu0.2).
qtbase5-dev is already the newest version (5.15.3+dfsg-2ubuntu0.2).
qtbase5-dev-tools is already the newest version (5.15.3+dfsg-2ubuntu0.2).
The following additional packages will be installed:
  python3-wheel
The following NEW packages will be installed:
  python3-pip python3-wheel
0 upgraded, 2 newly installed, 0 to remove and 7 not upgraded.
Need to get 1,337 kB of archives.
After this operation, 7,176 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu jammy-updates/universe amd64
python3-wheel all 0.37.1-2ubuntu0.22.04.1 [32.0 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu jammy-updates/universe amd64
python3-pip all 22.0.2+dfsg-1ubuntu0.2 [1,305 kB]
Fetched 1,337 kB in 3s (400 kB/s)
Selecting previously unselected package python3-wheel.
(Reading database ... 218086 files and directories currently installed.)
Preparing to unpack .../python3-wheel_0.37.1-2ubuntu0.22.04.1_all.deb ...
Unpacking python3-wheel (0.37.1-2ubuntu0.22.04.1) ...
Selecting previously unselected package python3-pip.
Preparing to unpack .../python3-pip_22.0.2+dfsg-1ubuntu0.2_all.deb ...
Unpacking python3-pip (22.0.2+dfsg-1ubuntu0.2) ...
Setting up python3-wheel (0.37.1-2ubuntu0.22.04.1) ...
Setting up python3-pip (22.0.2+dfsg-1ubuntu0.2) ...
Processing triggers for man-db (2.10.2-1) ...
Requirement already satisfied: pip in ./oak1/lib/python3.10/site-packages
(22.0.2)
Collecting pip
  Using cached pip-23.1-py3-none-any.whl (2.1 MB)
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.2
    Uninstalling pip-22.0.2:
ERROR: Could not install packages due to an OSError: [Errno 13] Permission
denied: '/home/dorte/oak1/bin/pip'
Check the permissions.

Failed installing dependencies. Could be a bug in the installer or
unsupported platform. Open a bug report over at
https://github.com/luxonis/depthai - exited with status 1 at line 120

I have all permissions as admin on this laptop...
Any advice?
Doerte

DDopfer, 2023-04-19:
Petr, never mind, I found the solution.
All is well.
Sorry for the bother