luxonis/depthai #1009: [BUG] Empty / Black infrared image

studentWUR, 2023-04-28:
**Describe the bug**
All of my camera's are running as expected, but for one of them the rectified right infrared image, is completely black.
As visible in the image below 
[image]
It seems that something is wrong with the IR right camera, but no errors appears.

I succesfully applied a factory reset using the [device manager](https://github.com/luxonis/depthai-python/tree/main/utilities), but the this did not help. 
Bootloader version: 0.0.24
Depthai version: 2.21.2.0

Any solutions for this problem? Or is this an hardware problem?

Thanks in advance.

themarpe, 2023-05-02:
This might be a hardware issue CC: @Erol444 

Did you try using cam_test utility (under depthai-python) as well?

studentWUR, 2023-05-08:
Sorry for my late reply I was on vacation. Yes this happens at cam_test as well. What should I do?

Erol444, 2023-05-08:
@studentWUR could you share the full log from cam_test, and perhaps [enable debugging](https://docs.luxonis.com/projects/api/en/latest/tutorials/debugging/)?

studentWUR, 2023-05-08:
Please let me know if this is sufficient

[log.txt](https://github.com/luxonis/depthai/files/11420398/log.txt)

Erol444, 2023-05-08:
Hi @studentWUR , From the log it looks like it's working as expected, so that you are receiving about 14 FPS from all connected cameras. Could you confirm?
```
Connected cameras:
 -socket RGB   : IMX378 4056 x 3040 focus:auto  - COLOR
 -socket LEFT  : OV9282 1280 x  800 focus:fixed - MONO
 -socket RIGHT : OV9282 1280 x  800 focus:fixed - MONO
USB speed: UNKNOWN
IR drivers: [('LM3644', 2, 99)]
Cam:      rgb           left          right         camd     [host | capture timestamp]
FPS:  13.67| 21.76  13.40| 26.67  14.16| 26.67   0.00|  0.00
```

studentWUR, 2023-05-09:
Yes, all of them have ~14 FPS.

Erol444, 2023-05-09:
@studentWUR so images are being streamed, they are just black? If that's the case, please send an email to support@luxonis.com and we will send you a replacement camera, it might be a hardware issue (likely connection of the CCMs to the PCBA).

studentWUR, 2023-05-09:
Yes, only for the right IR camera. I will contat support and close this issue. Thank you for your help