luxonis/depthai #1019: [BUG] {No extrinsic calibration available for OAK-D-S2}

chang-1, 2023-05-12:
I want to run the depthai_demo.py script, but I get an error message saying there is no extrinsic calibration data. Do I strictly need to run the board calibration, or does the repository contain any file or .json with the OAK-D-S2 parameters? Thanks!

Camera: OAK-D-S2
depthai version: 2.21.2

```console
python depthai_demo.py 

Depthai version installed:  2.21.2.0
Setting up demo...
Available devices:
[0] 1944301061C31A1300 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Warning: calibration data missing, using OAK-D defaults
[1944301061C31A1300] [1.2] [1.495] [StereoDepth(9)] [error] There is no available extrinsic calibration between camera ID: '1' and '2'
Stopping demo...

```

jakaskerl, 2023-05-13:
Hi @chang-1,
the default calibration data is available [here](https://github.com/luxonis/depthai-boards/blob/e9dfca0572a264cc93cb556c926e73a2609cdcae/boards/OAK-D-S2.json). 
AFAIK the warning shouldn't cause problems.

Jason-xy, 2023-08-29:
I have encountered the same issue when I run ` python3 depthai_demo.py`.
The output of my terminal:

```
Warning: calibration data missing, using OAK-D defaults
[19443010D138E01200] [1.1] [22.491] [StereoDepth(7)] [error] There is no available extrinsic calibration between camera ID: '1' and '2'
```

The versions of my software env are below:

```
depthai                       2.21.2.0            
depthai-sdk                   1.9.4 
```

vasup86, 2024-04-21:
Hey, has anyone found a solution for this bug? I'm getting the same error, 'There is no available extrinsic calibration between camera ID: '1' and '2'

vasup86, 2024-04-27:
Hey @jakaskerl, 
Yes, its OAK-D S2, I'm using version 2.25.0.0. This is the output when I run the `calibration_dump.py`,

`Is EEPROM available: True
User calibration: {
  "batchName": "",
  "batchTime": 1661229323,     
  "boardConf": "nIR-C00M00-00",
  "boardCustom": "",
  "boardName": "DM9098",       
  "boardOptions": 4,
  "boardRev": "R6M2E6",
  "cameraData": [],
  "deviceName": "",
  "hardwareConf": "F1-FV00-BC001",
  "housingExtrinsics": {
    "rotationMatrix": [],
    "specTranslation": {
      "x": 0.0,
      "y": 0.0,
      "z": 0.0
    },
    "toCameraSocket": -1,
    "translation": {
      "x": 0.0,
      "y": 0.0,
      "z": 0.0
    }
  },
  "imuExtrinsics": {
    "rotationMatrix": [
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ],
      [
        0.0,
        0.0,
        0.0
      ]
    ],
    "specTranslation": {
      "x": 0.0,
      "y": 0.0,
      "z": 0.0
    },
    "toCameraSocket": -1,
    "translation": {
      "x": 0.0,
      "y": 0.0,
      "z": 0.0
    }
  },
  "miscellaneousData": [],
  "productName": "OAK-D S2 FF",
  "stereoEnableDistortionCorrection": false,
  "stereoRectificationData": {
    "leftCameraSocket": 255,
    "rectifiedRotationLeft": [
      [
        0.0,
        0.0,
        0.0
      ],
[... 143 lines cut ...]
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 79, 65, 75, 45, 68, 32, 83, 50, 32, 70, 70, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 110, 73, 82, 45, 67, 48, 48, 77, 48, 48, 45, 48, 48, 0, 0, 0, 70, 49, 45, 70, 86, 48, 48, 45, 66, 67, 48, 48, 49, 0, 
[... cut ...]
0, 0, 0, 77, 97, 120, 119, 101, 108, 108, 0, 0, 0, 0, 0, 0, 0, 0, 0, 11, 89, 4, 99, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4, 0, 0, 0, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 
255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255, 255]`

jakaskerl, 2024-04-29:
Hi @vasup86 
Looks like the device is missing calibration. Please recalibrate it using: https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/calibration/. Sorry for the inconvenience. 

Thanks

flabrosse, 2024-07-08:
I just received a replacement camera (the previous one had the stereo cameras very badly aligned).  The new one is missing its calibration.  Not a very positive experience so far...