luxonis/depthai #1029: [BUG] AttributeError: 'TrackingConfig' object has no attribute 'speed'

maxsitt, 2023-05-20:
**Describe the bug**
When trying to run the [SDK object tracking example](https://github.com/luxonis/depthai/blob/main/depthai_sdk/examples/NNComponent/object_tracking.py), I get the following error as soon as an object is detected:

```
Traceback (most recent call last):
  File "/home/pi/object_tracking.py", line 22, in <module>
    oak.start(blocking=True)
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_camera.py", line 353, in start
    self.poll()
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_camera.py", line 381, in poll
    self._oak.check_sync()
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_device.py", line 44, in check_sync
    sync.check_queue(block=False)  # Don't block!
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_outputs/xout/xout_base.py", line 91, in check_queue
    self.on_callback(packet)
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_outputs/xout/xout_tracker.py", line 78, in on_callback
    self._add_tracklet_visualization(packet, spatial_points, tracklet2speed)
  File "/home/pi/sdkvenv/lib/python3.9/site-packages/depthai_sdk/oak_outputs/xout/xout_tracker.py", line 119, in _add_tracklet_visualization
    if self._visualizer.config.tracking.speed and tracklet.id in tracklet2speed:
AttributeError: 'TrackingConfig' object has no attribute 'speed'
```

**Minimal Reproducible Example**
https://github.com/luxonis/depthai/blob/main/depthai_sdk/examples/NNComponent/object_tracking.py

**Expected behavior**
Show object tracker output on frame.

**Attach system log**
{
    "architecture": "64bit ELF",
    "machine": "aarch64",
    "platform": "Linux-6.1.21-v8+-aarch64-with-glibc2.31",
    "processor": "",
    "python_build": "default Feb 28 2021 17:03:44",
    "python_compiler": "GCC 10.2.1 20210110",
    "python_implementation": "CPython",
    "python_version": "3.9.2",
    "release": "6.1.21-v8+",
    "system": "Linux",
    "version": "#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023",
    "win32_ver": "",
    "packages": [
        "blobconverter==1.4.1",
        "boto3==1.26.137",
        "botocore==1.29.137",
        "certifi==2023.5.7",
        "charset-normalizer==3.1.0",
        "depthai==2.21.2.0",
        "depthai-pipeline-graph==0.0.5",
        "depthai-sdk==1.10.1",
        "idna==3.4",
        "jmespath==1.0.1",
        "marshmallow==3.17.0",
        "numpy==1.24.3",
        "opencv-contrib-python==4.7.0.72",
        "packaging==23.1",
        "pip==20.3.4",
        "pkg-resources==0.0.0",
        "python-dateutil==2.8.2",
        "pytube==15.0.0",
        "PyTurboJPEG==1.6.4",
        "PyYAML==6.0",
        "Qt.py==1.3.8",
        "requests==2.30.0",
        "s3transfer==0.6.1",
        "sentry-sdk==1.21.0",
        "setuptools==44.1.1",
        "six==1.16.0",
        "types-PySide2==5.15.2.1.5",
        "urllib3==1.26.15",
        "xmltodict==0.13.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux rpi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64"
    ]
}

Thanks for looking into it!

themarpe, 2023-05-26:
CC: @daniilpastukhov  - please check this out

jakaskerl, 2023-06-13:
Broken in commit: [e8da50524d1ec7ae2e17c9e8df0987b8b4d5ff21](https://github.com/luxonis/depthai/commit/e8da50524d1ec7ae2e17c9e8df0987b8b4d5ff21)